MatterHackers/MatterControl #178: Only slicing and generationg gcode up about 10 layers into the print, then nothing

johnoly99, 2015-02-12:
Here's a funny one.  I can slice this model with the settings in the zip file linked below, and it only goes up about 10 layers or so and then just stops.  The 3d preview won't even load either?

Here's the link to the files
https://onedrive.live.com/redir?resid=145B2A0BC653974A%211260

It's MatterControl seemecnc v1.2.2 fyi

johnoly99, 2015-02-12:
So Lars, you'll appreciate this one.  I found that the settings profile in that zip has support checked ON.  When i unchecked the support box, it slices and finishes just fine, so somewhere in the support (which is generally awesome!) it is hanging up and failing

larsbrubaker, 2015-02-12:
Thank you John,

This was a good one :). I fixed it in your 'alpha' build, and it will go out with the next update.  The next update is just going to be a maintenance release and should not be more than a week or two out.

closed 2015-02-12 by larsbrubaker