MatterHackers/MatterControl #184: Cancel doesn't.

geneb, 2015-02-20:
This applies to build 1.2.2.3896.
When you click the Print button, clicking Cancel doesn't abort the heat-up process.  In order to abort the print, you need to disconnect and reconnect the printer (via the Disconnect button).
I don't know if it properly aborts during printing.

whosawhatsis, 2015-02-20:
This is because the firmware stops accepting commands while M109/190 is blocking. See discussion here: https://github.com/MarlinFirmware/Marlin/issues/685

The solution that has been proposed is for hosts (like MatterControl) to not send those codes, and instead replace them with M104/M140, followed by waiting for the temperature to reach the setpoint before sending additional commands. When doing this, the host should really still send the M109/190 after the temperature has been reached to ensure that the temperature stabilizes properly before proceeding.

geneb, 2015-02-20:
This behavior is the same with Repetier-Firmware 0.91.  Thanks for the info!
I should note that the Cancel function does work properly once the printer is actually working on the part.

larsbrubaker, 2015-02-21:
We do have a feature in our database to implement the M104 M140 tracking in software. So, hopefully it will rise to the top of our list soon. (Anytime users point things out it gives a big boos to a bug). Thank you.

closed 2015-02-21 by larsbrubaker