MatterHackers/MatterControl #190: Battery in Mattercontrol tablet?

whosawhatsis, 2015-02-27:
Sorry if this is the wrong place to ask, but does the Mattercontrol tablet have a battery, or is it only supposed to work while plugged in? I thought I saw an 80% battery indicator when I first turned it on, but now it shows 100% but loses power as if there was no battery when I unplug it. Also, it gets hotter than I would expect sitting plugged in, but doing nothing. I'm wondering if there's something wrong with the power circuit or (if it has one) the battery that I should be concerned about.

larsbrubaker, 2015-02-28:
It does not have a battery. The initial units that we got could not power the USB OTG at the same time they were plugged in so we had to have them rewired. It is something we are working on with our manufacturer.

closed 2015-02-28 by larsbrubaker

whosawhatsis, 2015-02-28:
Well, it's comforting to know that that excessive heat is not coming from a LiPo battery, but it's still a little disturbing. Is it perhaps using a linear regulator instead of a more efficient voltage converter? It could just be the processor, but it would be surprising for the processor to get so hot while doing nothing but displaying a static lock screen.