MatterHackers/MatterControl #200: Canceling an Insert operation leaves the 3D Viewer in edit mode.

geneb, 2015-03-11:
If you click Insert to add an additional file, clicking cancel dismisses the file dialog, but MC falls back to "edit" mode instead of reverting to the state it was in when Insert was clicked.  If this is intentional behavior, feel free to ignore me. :)

g.

larsbrubaker, 2015-03-12:
It would be better to enter edit mode on the confirmation of a selected file rather than the prompt. Good catch. I'll add it to our list. But to be honest it is not very high priority :).

geneb, 2015-03-12:
If I could make heads or tails of the code, I'd submit a patch.  You guys and your home-grown widget sets. :grinning: 
Thanks Lars!

g.

larsbrubaker, 2016-04-30:
So this is going to be fixed eventually by having the edit mode be always active (planed for the release after the next, so 1.7). Then not inserting will not change anything. Closing this as the feature is tracked elsewhere.

The new 3D view has much better inserting and much more editing features.

closed 2016-04-30 by larsbrubaker