MatterHackers/MatterControl #201: Inserting an object doesn't auto-arrange placement.

geneb, 2015-03-11:
I started with a simple cube (the default cube that's cooked into the 1.2.2 installer) in the center of the bed and then inserted a 75% scaled version of ORIONKEY.STL (http://repables.com/r/151/).  It placed the newly loaded STL file WAY off in space.  In order to see where the new part was loaded, I had to zoom out to such an extent that the virtual bed was very small.  (see attached image)
[image]

Clicking the auto-arrange button properly positioned both objects.

An inexperienced user may not know to zoom out to go looking for the object or the need to hit auto-arrange to get it on to the virtual build platform.  

I'm on the fence about this being a bug or a feature request. :)

g.

larsbrubaker, 2015-03-12:
Fixed! 

This will be in the next build. Thanks for the feedback. :)

closed 2015-03-12 by larsbrubaker

geneb, 2015-03-12:
\o/

sworah, 2015-04-15:
Request to make this an option for the user, sometimes I DO NOT want to rearrange everything upon insertion of a part.

larsbrubaker, 2015-04-15:
It only moves the newly inserted part into the build area. It does not re-arrange the other parts.