MatterHackers/MatterControl #202: Clicking Align will "stack" objects.

geneb, 2015-03-11:
Starting off with three objects:
[image]
Click Align and you end up with this:
[image]

Clicking Ungroup allows the two cubes to be un-stacked from one another.
Build is 1.2.2.3896.

larsbrubaker, 2015-03-12:
It aligns them to their original modeled coordinates. So, if you but a copy of a part on the same plate and align they will go into each other. I really want to improve this and make it clearer to the user what is going to happen and why.

closed 2015-03-12 by larsbrubaker

geneb, 2015-03-12:
How about change their color if two or more objects have the same origin point?