MatterHackers/MatterControl #204: 2 Issues in same print.(Round objects with flat sides and missing external perimeter.)

SpencerCWatson, 2015-03-16:
Trying to print shift knob for a car and there are 2 issues with the print

1. Round objects end up with 2 flat spots on opposite sides at mid point. ( Pictures).

2. When it is time to start the external perimeter for the part of the knob where the numbers are it does not extrude for 1/4 or more of the external perimeter. It goes through the motion for the external perimeter but no plastic is laid down. It shows up correctly in layer view but it has no extrusion when printing. Happens at about layer 188 ( Pictures)

Here is the link to my google drive with settings, gcode, stl, etc. Hopefully this can get fixed decently fast because I see these as serious bugs that really prevent a perfect print otherwise.
https://drive.google.com/folderview?id=0BwItvBIsxpO9djBSU3BydDBRZWs&usp=sharing

This happens with multiple objects of different size with different numbering. It also happens on multiple machines. Both issues are very repeatable.
[image]
[image]
[image]
[image]

whosawhatsis, 2015-03-16:
The flat spots are almost certainly a mechanical issue. Backlash in whichever axis moves perpendicular to them.

The other issue means that you need to tune your retraction settings (you may also be running too hot, which would make retraction more important). Should also get rid of all those pimples if you do it right. Unless MatterControl is failing to generate retractions when it's supposed to, neither of these is a software problem.

SpencerCWatson, 2015-03-16:
That flat spots are not a mechanical issue because they show up in the layer veiw as flat spots. If I rotate the object the flat spots follow the rotation. It also appears in my other machines and with other round objects. 

I was chasing down that crack issue as cracked in the part so I was trying a few different things. If it is my retraction that is an issue that would mean that my retraction is wrong for my other 2 machines as well and I find it odd that it only happens at the base of the very first number. Maybe I will dissable retraction all together and try again.

peter1960, 2015-03-16:
If the flat are showing in the view, have you tried one of the other slicers as you have a choice of 3.
As for the crack issue, is it a crack or when the layer was done at that point it did not stick to the previous layer as that was the start of the perimiter loop and there was too little filament extruded due to the retraction, if so at that point if you watch it you will see it start but the extrusion will cut the corner till enough filament is coming out to bond to the previous layer

whosawhatsis, 2015-03-16:
If the flats are showing up in the gcode view and follow rotation of the model, then there is an issue with your model. If they're not visible in the model, it's likely that your model isn't manifold, though I'd have to see the model to tell you more.

SpencerCWatson, 2015-03-17:
The model is in the link to my google drive in first post. It slices correctly in other slicers according to layer view but I will try my first print with another slicer to make sure. Peter I thought it was a crack at first so I was playing with my setting heavily to try to correct it but after watching it while printing I can see that the nozzle moves to the external part of the model where the R is then starts to move counter clockwise doing the perimeter. The Extruder motor is moving but it just takes a while for filament to start. There should be a retraction before the external perimeter but what I can't tell is if it is recovering from the retraction or not. It prints fine at that spot with retraction off. It seems like it retracts then does not unretract, it just starts trying to lay down filament but some of it is still retracted. All my other prints do not show this this behavior just this one in that spot.

peter1960, 2015-03-17:
I would think you have too much retraction, not sure what you are using for that but I normally leave default, do you even need retraction for this? I would think it's not crossing perimeters so you do not need it. Have you tried another slicer yet?

SpencerCWatson, 2015-03-17:
Using slicr 1.2.6 with all the same exact settings gets rid of flat spots and that missing part of external layer. I copied the setting over one at a time from matterslice to slicr then ran the exported gcode from slicr in mattercontrol. The only difference was infill type because slicr does not use triangles. Disabling retraction in matterslice gets rid of gap but the flat spots remain.

larsbrubaker, 2015-03-17:
@SpencerCWatson, I downloaded your model and sliced it with MatterSlice. I am not seeing any flat spots or errors in the gcode. I am also not seeing any unmatched retractions or unretractions. I have not printed the part (yet) but if you saw the flat parts in the gocde and I'm not than that doesn't seem to be the first problem. Can you send me your slice settings (to support@matterhackers.com)?  I would love to fix whatever is causing this problem.

[image]

I've attached an image of the gcode in case it looks different than yours.

SpencerCWatson, 2015-03-17:
larsbrubaker I sent you an email. I forgot to say that the object I uploaded needs rotated so that the numbers face up. Here is a picture of MatterSlice showing the flat spots and Slicr not showing any at the same layer.
[image]
[image]

larsbrubaker, 2015-03-18:
@SpencerCWatson, thanks for hanging in there with me. I have a possible solution to this problem. The CleanPolygon code in the clipper library seems to be being too aggressive (or I don't understand it correctly). I'll try and make a new experimental build tomorrow that you can test if you would like.

SpencerCWatson, 2015-03-18:
That would be great. I will do whatever I can to help get this fixed.

larsbrubaker, 2015-03-18:
[Here](https://mattercontrol.appspot.com/downloads/development/ag9zfm1hdHRlcmNvbnRyb2xyQQsSB1Byb2plY3QYgICA0KOn2goMCxINUHVibGljUmVsZWFzZRiAgICA-q6cCgwLEgZVcGxvYWQYgICAgIaImgoM) is a link to an experimental build that fixes the flats problem. It still seems to be missing a layer near the top (not generating the correct outline). I'll look into that more.

SpencerCWatson, 2015-03-19:
@larsbrubaker  That build did fix the flat spots and it also got rid of the missing perimeter issue at least for the 1 test print I was able to make today. I did notice though during this print that my machine sounded like it never has before when doing external and internal perimeters near the bottom of the print, so small circles. it sounded and looked kind of jagged. It did not have that smooth stepper motor sound and I could also see what looked like the nozzle almost skipping around the small perimeters. its motion was not fluid. Maybe what was changed in this build was a little over corrected making too many movements out of the perimeters? I will try to print again tomorrow from the SD card to make sure this was not my computer. Thanks for the quick fix.

SpencerCWatson, 2015-03-19:
@larsbrubaker Alright 3 prints later and here is what I know. The flat spots are fixed and the missing perimeter is gone.(Not sure if you did anything with the perimeter though) It seems like the GCODE is more than twice as large now and printing from my computer the print does stutter during perimeters. Printing from SD card it is smooth. My computer specs are i7 2.8-3.8GHz processor, 8gb of ram, SSD.

larsbrubaker, 2015-03-19:
@SpencerCWatson So, the CleanPolygons function in a library that we use (clipper) does not handle col-linear points correctly. The little hack that I tossed into experimental causes way too many segments to be created. I have a new bit of code that is close to working (has a few edge cases I need to solve) that is maintaining great curves and reducing points correctly. I hope to have a new build today that fixes this completely.

[image]

larsbrubaker, 2015-03-19:
Ok, so [here](https://mattercontrol.appspot.com/downloads/development/ag9zfm1hdHRlcmNvbnRyb2xyQQsSB1Byb2plY3QYgICA0KOn2goMCxINUHVibGljUmVsZWFzZRiAgICA-q6cCgwLEgZVcGxvYWQYgICAgIbshwoM) is the link to the new experimental build that should be the final fix for this issue :). Please let me know if you find more bad stuff. And thanks for helping make this project better.

The final analysis of this was that this model  was exposing a very subtle bug in clipper. If there were a very large number of edge points on a relatively shallow curve, one can't just merge each as they walk the curve in order or they can collapse curves that have small enough inter point deltas but a large total curve.

Imagine a very tessellated circle. As we walk the edge, each triple we find can be collinear within our error and so we remove the middle point and move on to the next point. The new middle point is very close to the new third point and so we collapse and remove the second point. We can remove and create a large flat on the edge of the circle.

Hope that all makes sense.

larsbrubaker, 2015-03-25:
Put that build through testing and fixed one issue. This code is now final and will be out in the next release. Thanks for the help. [Here](https://mattercontrol.appspot.com/downloads/development/ag9zfm1hdHRlcmNvbnRyb2xyQQsSB1Byb2plY3QYgICA0KOn2goMCxINUHVibGljUmVsZWFzZRiAgICA-q6cCgwLEgZVcGxvYWQYgICAgIb6lgoM) is the experimental build with the final code in it.

closed 2015-03-25 by larsbrubaker