MatterHackers/MatterControl #206: Cancel while preparing to print doesn't immediately cancel.

ChrisWeiss, 2015-03-18:
Not sure if it's waiting for pre-heating to finish, but hitting Cancel doesn't seem to cancel the print if pressed between the time the user has clicked Print and the actual printing starts.

o3dodgesrt4, 2015-06-26:
iv noticed this too...even during prints that have been running for a while, when you press cancel it keeps printing for about 1 minute longer then quits...

whosawhatsis, 2015-06-26:
It's a firmware issue. While waiting to reach temperature, the printer keeps sending updates on its temperature, but pretty much stops responding to anything else short of a hard reset, and some printers don't even make that possible over USB.

larsbrubaker, 2016-05-26:
this has been fixed as of 1.5. We now issue a M104 when an M109 is requested and wait in software.

closed 2016-05-26 by larsbrubaker