MatterHackers/MatterControl #217: [Feature Request] Allow the user to arrange order of queue

sworah, 2015-03-23:
You cannot move items around in the queue order, you are stuck with the order you added them. There should be a method to drag them around or something.

annafeldman, 2015-08-04:
Submitted pull request that adds a "prioritize" button to mouseover on each item that bumps an item up in the queue

https://github.com/MatterHackers/MatterControl/pull/349

whosawhatsis, 2015-08-04:
I'd rather be able to drag queue items to put them in a new order. This would be more consistent with UI elements in other programs.

For the tablet version, there should be boxes (usually with three horizontal lines) on the end of each item that can be used to touch and drag each item. Clicking "edit" is sometimes required to make such boxes available.

annafeldman, 2015-08-04:
The pull request was submitted in part to solicit further feedback on how to fit with the UI conventions; I didn't want to create a fancy UI for changing the order that would just have to be rewritten!

The other option I considered is having the edit window have an arrow next to it that let you move items up and down, however I felt that stacking functionality in there was not ideal as it's a bit difficult to find and unintuitive.

As such, I agree that dragging is a better solution. One way to keep it consistent between tablet and desktop might be to have a grab texture appear on both that you use to change the order.

stuartpb, 2017-12-24:
#349 was closed saying that this has been put on hold until the queue can be drag-and-dropped; how close is that?

larsbrubaker, 2017-12-24:
MatterControl 2.0 is due out in March. We'll likely have alphas before then. The library has been redone and is much more usable. You can look at the experimental build now it is here https://mattercontrol.appspot.com/downloads/matterhackers-experimental-windows/release