MatterHackers/MatterControl #257: Gcode output menu is unusable

whosawhatsis, 2015-05-14:
In 1.2.2.4257 (and probably other versions), at least on the Mac, the gcode output menu is unusable because the menu pops up within the window (clips off at the edge), but it is at the bottom of the window and does not allow scrolling, so the all but the first two menu items are inaccessible unless the window is extremely large (larger than my screen will allow at 1680x1050). Ideally, this should be solved by using the systems's native interface elements, which I'm sure would solve other issues as well, but otherwise the elements used need to be smart enough to notice when the whole menu won't be visible and adjust to fix that.

[image]

jlewin, 2015-05-14:
Thanks for the report. This behavior was recently fixed and will be included in the next release

[image]

unlimitedbacon, 2016-05-12:
@jlewin, can you close this?

johnlewin, 2016-05-12:
Indeed

closed 2016-05-12 by johnlewin