MatterHackers/MatterControl #283: Building on Gentoo

au-phiware, 2015-06-07:
Hello,

I have a Kossel Pro and am attempting to build MatterControl on my gentoo system by following the [wiki](http://wiki.mattercontrol.com/Building_MatterControl).

In brief I have done the following:

    emerge dev-util/monodevelop
    emerge =dev-lang/mono-3.2.8
    mozroots --import --sync
    git clone --recursive https://github.com/MatterHackers/MatterControl.git
    git clone https://github.com/MatterHackers/MatterControlPlugins.git
    mono MatterControl/.nuget/NuGet.exe restore MatterControl/MatterControl.sln
    mdtool build -c:Release MatterControl/MatterControl.sln

At this point I encounter this curious error:

    MonoDevelop Build Tool
    File 'MatterControl/MatterControl.sln' is not a project or solution.

There appears to be nothing wrong with the `MatterControl.sln` file (indeed NuGet didn't complain), git doesn't report any problems. I checked out the 1.2.3 branch and updated the submodules but still, I get the same error.

Please advise.

au-phiware, 2015-06-07:
Some of my details:

````
$ uname -a
Linux fibonacci 3.19.3-stable-ath10k #42 SMP Sun May 24 14:15:40 AEST 2015 x86_64 Intel(R) Core(TM) i7-4710HQ CPU @ 2.50GHz GenuineIntel GNU/Linux
$ mono --version
Mono JIT compiler version 3.2.8 ((no/b1e8599 Sun Jun  7 23:43:29 AEST 2015)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       altstack
        Notifications: epoll
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug
        LLVM:          supported, not enabled.
        GC:            sgen
````

jlewin, 2015-06-07:
It is strange the NuGet had no problems with the .sln file and mdtool does. I'd suggest compiling with xbuild as a variation to rule out a quirk in mdtool. In addition, have you verified with git that the .sln is unchange from the HEAD commit?

xbuild MatterControl/MatterControl.sln

au-phiware, 2015-06-07:
Xbuild has no complaints regarding the sln file, but does produce build errors: https://bpaste.net/show/fe6ad790f639

My working copy is clean:

````
$  git status; git submodule; git -C Submodules/agg-sharp status; git -C Submodules/MatterSlice status
HEAD detached at origin/1.2.3
nothing to commit, working directory clean
 1466ab81ccc5786e1737d95005e8aa7db22dc017 Submodules/MatterSlice (remotes/origin/1.2.3)
 c6b7b43313bb2f899277066c8a89c48b70ecf42b Submodules/agg-sharp (remotes/origin/1.2.3)
HEAD detached at c6b7b43
nothing to commit, working directory clean
HEAD detached at 1466ab8
nothing to commit, working directory clean
````

Should I proceed with xbuild or attempt to fix the problems with mdtool.

au-phiware, 2015-06-08:
I'm just looking through that list of errors... the first thing that jumps out at me is that there's a typo in the file path `SubModules\agg-sharp\Tesselate\Tesselate.csproj` (the actual file path uses a lowercase 'm')...

jlewin, 2015-06-08:
>Should I proceed with xbuild or attempt to fix the problems with mdtool.
We use both. I often build with xbuild at the command line, monodevelop/mdtool in the gui and the build servers use mdtool. If mdtool is broken on Gentoo it seems reasonable to swap it out for xbuild.

>I assume I don't need the other projects that are missing:
That's correct. The project should build and function without the additional plugins

au-phiware, 2015-06-08:
I have created a set of ebuilds for gentoo's portage (packaging) system: https://github.com/au-phiware/gentoo-overlay

closed 2015-06-10 by au-phiware