MatterHackers/MatterControl #287: ignoring first layer extrusion width for infill line spacing

whosawhatsis, 2015-06-09:
In 1.2.3.4619, the first layer extrusion width setting is being ignored for line spacing of the infill, but not for extrusion volume, resulting in way too much plastic being laid down. The spacing of the perimeters is correct, but the infill lines are too close together.

[image]
First layer extrusion width = 100%

[image]
First layer extrusion width = 250%

Note that this makes it impossible to print anything when this value is not set to 100%, because the lines are still being extruded at 250% width, even though the spacing between them is only sufficient for a 100% width extrusion.

larsbrubaker, 2015-06-09:
Thank you. We'll get this fixed right away. There is going to be a small bug fix release very soon and this will be in it.

larsbrubaker, 2015-06-09:
Fixed in the next build.

closed 2015-06-09 by larsbrubaker