MatterHackers/MatterControl #290: Unhandled Exception: System.NullReferenceException at OpenTK.Graphics.GraphicsMode.get_Default

au-phiware, 2015-06-10:
Hello,

I have successfully built and run MatterControl under gentoo/linux on my laptop. I am now attempting to run it on a smaller netbook (Acer Aspire One, also gentoo) and I am getting the exception below. I suspect that it has either something to do with my graphics card (I've only got the generic x11 opengl drivers) or I'm missing some library dependency.

Any help would be greatly appreciated.

My graphics card is: Intel Atom Processor D2xxx/N2xxx Integrated Graphics Controller

Many Thanks,
Corin

````
Unhandled Exception:
System.NullReferenceException: Object reference not set to an instance of an object
  at OpenTK.Graphics.GraphicsMode.get_Default () [0x00000] in <filename unknown>:0 
  at OpenTK.GLControl..ctor () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.MyGLControl..ctor (Int32 bitDepth, Int32 setencilDepth) [0x00000] in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) MatterHackers.Agg.UI.MyGLControl:.ctor (int,int)
  at MatterHackers.Agg.UI.WindowsFormsOpenGL..ctor (MatterHackers.Agg.UI.AbstractOsMappingWidget app, MatterHackers.Agg.UI.SystemWindow childSystemWindow) [0x00000] in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) MatterHackers.Agg.UI.WindowsFormsOpenGL:.ctor (MatterHackers.Agg.UI.AbstractOsMappingWidget,MatterHackers.Agg.UI.SystemWindow)
  at MatterHackers.Agg.UI.WidgetForWindowsFormsOpenGL..ctor (MatterHackers.Agg.UI.SystemWindow childSystemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WindowsFormsOpenGLFactory.CreateSurface (MatterHackers.Agg.UI.SystemWindow childSystemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.OsMappingWidgetFactory.CreateOsMappingWidget (MatterHackers.Agg.UI.SystemWindow childSystemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.SystemWindowCreator_WindowsForms.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.get_Instance () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () [0x00000] in <filename unknown>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object
  at OpenTK.Graphics.GraphicsMode.get_Default () [0x00000] in <filename unknown>:0 
  at OpenTK.GLControl..ctor () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.MyGLControl..ctor (Int32 bitDepth, Int32 setencilDepth) [0x00000] in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) MatterHackers.Agg.UI.MyGLControl:.ctor (int,int)
  at MatterHackers.Agg.UI.WindowsFormsOpenGL..ctor (MatterHackers.Agg.UI.AbstractOsMappingWidget app, MatterHackers.Agg.UI.SystemWindow childSystemWindow) [0x00000] in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) MatterHackers.Agg.UI.WindowsFormsOpenGL:.ctor (MatterHackers.Agg.UI.AbstractOsMappingWidget,MatterHackers.Agg.UI.SystemWindow)
  at MatterHackers.Agg.UI.WidgetForWindowsFormsOpenGL..ctor (MatterHackers.Agg.UI.SystemWindow childSystemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WindowsFormsOpenGLFactory.CreateSurface (MatterHackers.Agg.UI.SystemWindow childSystemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.OsMappingWidgetFactory.CreateOsMappingWidget (MatterHackers.Agg.UI.SystemWindow childSystemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.SystemWindowCreator_WindowsForms.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.get_Instance () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () [0x00000] in <filename unknown>:0 
````

jlewin, 2015-06-15:
There's a note in the MyGLControl constructor that says:

// If you have an error here it is likely that you need to bulid your project with Platform Target x86

Since this error is being throw in the contructor it's worth checking... maybe verify the build configuration specified on the xbuild command line. You might also consider compiling a debug build to get line numbers in the callstack

au-phiware, 2015-06-30:
Thank you for the advice.

It took me a while to realise that I needed to pass `--debug` option to `mono` in order to get debugging information. You are right about the constructor: [output here](https://bpaste.net/show/a487cbe3aee7).

I attempted to compile with the x86 target but no exe was produced:

     xbuild /property:Platform=x86 MatterControl.sln
[output](https://bpaste.net/show/f2e8f4001eba)

I then attempted to compile just the project containing the offending code but xbuild complained about an invalid configuration:

    xbuild /property:Platform=x86 Submodules/agg-sharp/PlatformWin32/agg_platform_win32.sln
[output](https://bpaste.net/show/73cadecbbdb4)

(Note I ran a clean between each command.)

I have tried various different combinations of configuration (`{Debug,Release}{,64}`) and platform (`x86 x64 AnyCPU "Mixed Platorms"`). Most did not compile and needless to say, I haven't had much luck.

The NullReferenceException is produced within OpenTK... am I missing a dependency?

For the record, the above was from the 1.2.3 branch 52b3836, I also tried from master (fa67afa) with the same results.

In a last ditch attempt I defined the `USE_GLES` constant, which didn't work with MatterControl.sln so I tried just Submodules/agg-sharp/PlatformWin32/agg_platform_win32.sln but that complained about a missing project file, so I thought maybe I'll try Submodules/agg-sharp/PlatformWin32/PlatformWin32.csproj but that complains about OutputPath not being set, and I'm not sure what it's supposed to be set to... maybe Submodules/agg-sharp/PlatformWin32/obj? But alas, no that had the same problems as MatterControl.sln... which when I think about it it's not so surprising... I'm just shooting in the dark here...

:8ball: says outlook is gloomy

jlewin, 2015-07-01:
Thanks for giving that a shot. More research reveals that none of the projects are configured to build when the solution platform configuration is "x86" and ultimately the "Any CPU" solution platform is hard-coded in each project with x86 as the platform build target. Thus, we're already compiling with a platform target of x86 and nothing obvious explains the null reference.

One additional bit of insight would be useful. If you have time, can you run MatterControl with the MONO_LOG_LEVEL set to debug and upload the results for review? Something like the following should produce what we need:

<code>MONO_LOG_LEVEL=debug mono MatterControl/bin/Release/MatterControl.exe  > log.txt</code>

au-phiware, 2015-07-03:
That produces over 1300 lines: https://bpaste.net/show/2a373842e383

I'm guessing there's no need to worry about the missing shared object files... but it doesn't mean much to me...

au-phiware, 2015-10-04:
I have revisited this problem (using master and 1.4.0 branches) and found that the problem lies in the xrandr adapter/driver/interface/whatever of OpenTK.

When I execute `xrandr` I encounter the following message:

    xrandr: Failed to get size of gamma for output default

Thus, after executing the following command, to set the gamma of my display, I no longer encounter the NullReferenceException when opening MatterControl.

    xrandr --output default --gamma 0:0:0 --mode 1024x600

So, I have now installed that command in a xinitrc script. I have also reported this issue: opentk/opentk#297

closed 2015-10-04 by au-phiware