MatterHackers/MatterControl #305: Slicer file saving/compile issue

o3dodgesrt4, 2015-06-26:
after making a STL in 123d design by autodesk and adding it to the print que, i click on the layer view then generate to view my stl with the settings i want using the SLIC3R Slicer engine.. EVERYTIME i do that it hangs at "saving intermediate file" and crashes the matter control...however, if i run it thru SLIC3R slicer program itself and save as an overwrite to the STL and re-add to the print que, it works...odd bug?  when in matter slicer mode it works just fine but the resolution is much worse when printing... looks like a 10% infill with thick layers even tho its set to %50 and .20mmLH...

whosawhatsis, 2015-06-26:
The infill thing sounds like an old Slic3r bug related to the combine infill feature, and may be because MatterControl is using an old version of Slic3r.

Could also just be a difference in the profiles being used by the two versions of Slic3r. Try setting "infill every X layers" to 1 and see if that makes a difference.

o3dodgesrt4, 2015-06-26:
tried that.. made no difference...i redesigned the stl online at tinkercad as well, saved it and loaded it...used the same settings, same result...iv talked with them about using slic3r before and they said they have had issues with it as well...

johnlewin, 2016-05-26:
If this is still happening, please add the file and your settings and we will reopen this issue.

closed 2016-05-26 by johnlewin