MatterHackers/MatterControl #306: print leveling-more options

o3dodgesrt4, 2015-06-26:
the 3 point leveling works ok... however it doesnt work that great on aluminum heated beds that are thinner than 5mm... i tried a 3mm and a 4mm and both when heated went out of level and after comparing the configuration wizard settings to the readings of my dial gauge i noticed they were off by as much as .15mm... what i also have a issue with is that once im configured to be level and the bed heats up the center of the bed bulges by .05mm since it cant expand being fixed to the lower carriage... iv used a inductive sensor to get AB leveling thru marlin for comparison and i havent had an issue yet... is there any way to have options to do 4 or 5 point AB leveling ? i would think if you copy paste the code from the working 3point and add in the changes for the math it should work just fine... Marlin firmware is pre set up to do up to 9 points of measure, copy paste that and mod as needed...

tommck, 2016-02-10:
I'm using a delta printer and the software has 13 points as an option with the software leveling.  So, I am not sure this is an issue anymore

o3dodgesrt4, 2016-02-10:
Yeah im using the newest version...now has 3 7 and 13 point
leveling...works good too..

tommck, 2016-02-10:
You should close this then

o3dodgesrt4, 2016-02-10:
Okay

o3dodgesrt4, 2016-02-10:
closed....latest update now has multiple options for bed leveling

closed 2016-02-10 by o3dodgesrt4