MatterHackers/MatterControl #317: MatterControl hangs after "restart"

whosawhatsis, 2015-07-02:
On the Mac, using MatterControl version 1.3.0.4743, after switching the display mode between normal and touchscreen, clicking the restart button to reload the app with the new interface causes it to hang, requiring a force quit. After force quitting, it does come up with the new display mode, but switching back causes the same hang.

unlimitedbacon, 2017-01-25:
Tested this on macOS 10.10.5 with MC 1.6.2.8242. The app does not hang, however after closing itself it does not reopen. You must restart the app manually.

larsbrubaker, 2018-09-24:
Restart has been removed in 2.0, and the new Mac version is almost ready.

Closing.
Please file a new bug if this is still a problem.

closed 2018-09-24 by larsbrubaker