MatterHackers/MatterControl #318: MatterControl crash.

geneb, 2015-07-03:
When adding a new printer, MC falls over hard when trying to enter the name of the new printer - this is on a fresh install of the current SeeMeCNC build.

Data:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	MatterControl.exe
  Application Version:	1.2.2.0
  Application Timestamp:	54eb9495
  Fault Module Name:	StackHash_cc51
  Fault Module Version:	6.1.7601.18869
  Fault Module Timestamp:	55636317
  Exception Code:	c0000374
  Exception Offset:	000cea5f
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	cc51
  Additional Information 2:	cc516d0487867485162f0cda643a34b9
  Additional Information 3:	da70
  Additional Information 4:	da701d5ac9674df5063cf08e79329f91

g.

gregory-diaz, 2015-07-06:
Gene, 

I tried reproducing the problem several times this morning on both Windows and Mac builds of the SeeMeCNC versions of MatterControl and could not get MatterControl to crash. I also made sure to test the current 1.2 stable build, and the newer 1.3 version that should be pushed to stable very soon. Could you provide me with the following information so that I might have more success reproducing the problem: 

1. What operating system you are on.
2 .Any type of procedure that consistently results in the crash

We are looking into the problem and it will be fixed, but the more detailed information that we can get from you the quicker we can find a solution to the problem!

Thank you, 
Greg

geneb, 2015-07-06:
I was using a 19" Toshiba Satellite running Windows 7 (Home Ed. I think).  The system is current on service packs & patches.  Any time I tried adding a printer it died in the manner described.  If you like, I can make the machine available for you to remote in on an test directly via Go2Meeting or other remote desktop service.

I also have the option of pulling the code and running within Visual Studio (just let me know what version I should use and MSDN shall provide. :) ) to see if it'll die there.

g.

gregory-diaz, 2015-07-07:
Go to Meeting or JoinMe sounds like it could be the best way to go only because building the project from source through visual studio can be very time consuming and challenging. Feel free to email me at greg.diaz@matterhackers.com and maybe we can set up a remote desktop meeting

pugwng, 2016-01-21:
Nothing else? I`m facing the same problem when trying to add a printer in a fresh install.

johnlewin, 2016-05-26:
This should be fixed with new OpenGL allocation tracking as of 1.5.

closed 2016-05-26 by johnlewin