MatterHackers/MatterControl #319: MatterSlice is the only slice engine to choose

didli, 2015-07-04:
Hi !
I'm currently using MatterControlSetup-1.2.2.tar.gz with Fedora 22 x64, and since I can't installed it with apt package manager, I'm using it as a standalone program. It works fine as it is, but other engines currently installed (I have cura and slic3r) won't show up in the advanced tabs.
Is there anything I can try to have these choices ?
Thanks !

gregory-diaz, 2015-07-06:
Hello,

The version of MatterControl that you are using only includes MatterSlice as an available slicer. Slic3r and CuraEngine  are not  included in MatterControl 1.2.2, however if you switch to your Update Notification Feed to Beta and download then install the available update, you will see that slic3r is included with MatterSlice in MatterControl 1.3.0 for Linux.

didli, 2015-07-06:
Thank you ! I've just managed to get 1.3 installed yesterday, and yes, absolutely, I can now use slic3r in mattercontrol ^^. I wasn't aware it was a feature only accessible from update. Why Cura is missing ? Coming in a future update too ?
Either way, I'm glad to use MatterControl : it works pretty well for me and I'm still discovering amazing things with it (like the export to sdcard option).

unlimitedbacon, 2015-08-18:
There is a little more [information on the wiki](http://wiki.mattercontrol.com/Running_on_Linux) about getting Cura and Slic3r to work.

gregory-diaz, 2015-08-18:
In version 1.2.2 of MatterControl Linux we did not include Slic3r or CuraEngine in the package because of time constraints and build complications that we were encountering. The reason Slic3r is only available via update is because version 1.3 of MatterControl included Slic3r in the package while 1.2.2 did not. It is our goal to make all three slicers available to MatterControl linux users and hopefully we will add CuraEngine to the package in the next update. If you really want CuraEngine as a slicing option you can place a CuraEngine executable into the same place that the Slic3r folder is in the MatterControl package. I have not thoroughly tested whether we would need to modify code to get it to actually work but if you really want CuraEngine you can try that and see if MatterControl discovers it. If not hopefully we can get it into the next update. We are very happy that you are enjoying using MatterControl! Thank you!

johnlewin, 2016-05-26:
Fixed in 1.2.2

closed 2016-05-26 by johnlewin