MatterHackers/MatterControl #330: Mac Preferences Survive Application Delete

Jacobknew, 2015-07-18:
On a Mac, preferences persist after deleting the application. The ability to fully remove these and do a full "uninstall" would be very helpful.

As an option, just creating a wiki page explaining how to remove all application data would solve the problem for more capable users. 

I have tried looking for any preferences files etc. that seem related and deleting those, as well as spending about 15 minutes browsing through the source code, and I have not had any luck figuring out how to fully remove all of the settings (printers added, material settings, etc).

whosawhatsis, 2015-07-18:
A better solution would be for MatterControl to follow standard Mac conventions for storing preferences. I.e. either using a single plist file at ~/Library/Preferences/MatterControl.plist or using multiple files of the appropriate types in ~/Library/Application Support/MatterControl.

FWIW, I believe the Mono framework, by default, stores it in some indiscernible format, mixed in with the application data for other Mono apps. Very annoying.

whosawhatsis, 2015-07-18:
Actually, looks like MatterControl is using ~/.local/share/MatterControl. Moving this to ~/Library/Application Support/MatterControl would make it reasonably standards-compliant, by the looks of things. Most of the settings are in an SQLite database, which is not as bad as what I expected to find.

Jacobknew, 2015-07-18:
Ah, yes, that explains why I wasn't finding it, I was restricting my searches to ~/Library/ and /Library/. Yes, that would be ideal, though just knowing it is in ~/.local/share/ solves my immediate problem.

unlimitedbacon, 2015-08-18:
I added a [note to the wiki](http://wiki.mattercontrol.com/Frequently_Asked_Questions) describing where the application data is stored. Feel free to add any other information you think would be helpful.