MatterHackers/MatterControl #348: Bridge Speeds not being obeyed

johnoly99, 2015-08-04:
Hey guys.

Bridge speed isn't being obeyed on smaller objects, where it looks like the cooling threshold is kicking in and over riding it?  Would it be better practice to obey bridge speeds and settings for bridge moves?  I think so, otherwise you would never be able to calibrate bridging on smaller objects.  Here's some images for examples
[image]
[image]

gamecompiler, 2016-03-26:
I have the same issue, has it been fixed ?

unlimitedbacon, 2016-05-20:
There is a pending fix in the code now that prevents the bridge speed from being ignored completely, however it does not address the way that cooling affects the bridging speed.

jlewin, 2016-05-24:
Original fix from  Feb that restored bridging speeds: MatterHackers/MatterSlice@eb9c1a0a55c5baed44fc8267c6f82212be10bbd3

johnlewin, 2016-05-24:
Just committed some changes to MatterSlice that should resolve this issue. These updates will be available in 1.6+

closed 2016-05-24 by johnlewin