MatterHackers/MatterControl #370: Cannot Export as X3G

HarlemSquirrel, 2015-08-21:
I am using Ubuntu 15.04 and am trying to export as X3G for my Replicator 2. I followed the directions on here: http://wiki.mattercontrol.com/Export_Gcode_as_X3G but when I try to export it nothing is created and there is no error message. I can export as STL or gcode but of course this will not work with my Replicator 2.

gregory-diaz, 2015-08-21:
First of all, I am sorry that you are having difficulty with the Export to X3G feature at the moment and understand your frustration. However, we will be completely reworking our X3G implementation in our next MatterControl update. I think you will be pleasantly surprised by our new X3G support. The new update should be out in a month or so. Hang in there just a little bit longer. It will be worth the wait!

HarlemSquirrel, 2015-08-21:
Thank you for your reply. That is great news! I am really looking forward to an open source slicer for my Replicator 2. I use Linux and dabble with 3D printing but am not a fan of Makerbot's proprietary tools that are often not up to date. MatterControl looks very promising! I shall wait with bated breath.

HarlemSquirrel, 2015-11-17:
OK, I am now able to print to my Replicator 2 with MatterControl 1.4 on Ubuntu 14.04! However, I am still having the same issue exporting my files as x3g where I simply get no file and no error message.

gregory-diaz, 2015-11-18:
Thank you for your feedback it is much appreciated. I just tested our extremely old export as X3G feature and have confirmed that it no longer works on Ubuntu. The good news is that we are currently working on a newer and more robust implementation of our Export X3G feature that should be available in our next release!

HarlemSquirrel, 2015-11-19:
Awesome! I can't wait!

HarlemSquirrel, 2016-03-30:
OK, I can confirm that MC 1.5 now allows me to export as X3G! However, I seem to be having trouble actually printing with it on my Rep 2. So far, the printer goes through the motions but is maybe cm or so too high from the build plate and nothing is extruded. I can print via USB without this problem.

johnlewin, 2016-05-26:
This is fixed in 1.5.3. Thank you.

closed 2016-05-26 by johnlewin