MatterHackers/MatterControl #397: Settings aren't applied when export G-Code

julianrendell, 2015-09-16:
Using version 1.3 on OS X, I went through and updated the settings to match my printer- including setting the temperature to 235'C.

I sliced a model and exported the generated g-code, and then uploaded it to the Octoprint instance I use to drive my printer. The print started, but the speeds were way to fast for my setup.
Further checking the generated G-Code and I could see the temp was set to the default 215, not the value I had set.

I also noticed that G-Code output doesn't seem to be regenerated after changes and re-export. Even after deleting the old G-Code file, the new file is recreated with the same timestamp as before. The new file also does not seem to reflect the options I'd created. Is it using a cached copy?

I don't plug my printer into my computer so gcode-export is the only way I can use MatterControl. 

is this a bug, or something I've missed in how to export G-Code?

gregory-diaz, 2015-09-16:
Julian, 

I tried reproducing the bug that you are experiencing on both Mac and Windows and did not get behavior that you reported. I tested several variations of the following steps on each platform:

1. Exported gcode 
2. Changed extruder temperature and saved slice settings
3. Exported gcode again (did not rename the file)

The file was updated and the gcode in the file reflected the changes that were made in slice settings. 

Here is a brief explanation of how to export gcode:

1. Create and select a printer profile
2. Add a part to the queue
3. Select the part and click the 'Export' button at the top of the queue 
4. When the Export Item window opens, click the Export as gcode button
5. Select a file location to save to and click save

gmlewis, 2015-11-30:
I've seen the same thing as @julianrendell. I generated the gcode file from MatterControl, then printed the design, then went back and modified the design, downloaded a new STL file, and MatterControl generates the exact same gcode file it output before with the exact same timestamp and does this instantly without any "processing layer ..." steps.  It looks like it is caching the gcode file for that design. This is on a Mac. I couldn't find where the file is being cached. Very frustrating.

larsbrubaker, 2016-05-26:
This is fixed in 1.6 (next update)

closed 2016-05-26 by larsbrubaker