MatterHackers/MatterControl #398: Customized / manual support placement

Patola, 2015-09-17:
Simplify3D has it and CraftUnique Craftware has it too. You can also add it externally to the stl through meshmixer. Cura and slic3r are working to implement it, and I think MatterControl should too: instead of only allowing automatic generated support through parameters, the user should be able to add them directly to the part, by mouse.

The way this works in Simplify3D is: you have a dialog box which lets you add support towers using the cursor, remove support towers using the cursor, run the automatic support algorithm to generate the support towers in place according to the parameters set in the configuration and clear all supports in the current part. It colors the supports darker than the actual part so it's easy to distinguish.

If you have customized supports in place but the support option is off, it warns you if you really want this off, lets you select yes or no, then proceeds.

MadSc1, 2015-12-14:
Bump!   Major bump !  I would rank this as highly important and a feature which would have a major improvement to prints.

larsbrubaker, 2015-12-14:
We are working on a dramatically improved automatic support generator, this will then be extended to allow user editing of both ignore and force regions. The new code makes it much easier to add hint regions and also has much better interface between the support and the bottom layers.

Thanks for the feedback.