MatterHackers/MatterControl #419: corrupted eprom settings when switching between two machines

ebo, 2015-10-06:
I have been using to MatterControl-1.3.0 on both a Kossel-Pro and a MendleMax-1.5 from one of my Gentoo boxs.  While setting up the old Mendle, one of the guys suggested that I use MatterControl to to temporarily adjust the Mendle's eprom settings until I figure out its tuning and then reflash the controller.  None of the changes I was making to the Mendle's eprom settings in MatterControl seemed to work, so I went back and hand edited them in the firmware, and then when I switched back to the Kossel today it was not running correctly and was dog slow and had the wrong filament feed.  As far as I can tell, the database for the eprom settings of the two machines are corrupted with each others settings.  I am going to wipe the settings and run each printer on a separate laptop for now, but it would be nice to figure out how to run multiple machines (either one at a time or possibly even two instances -- ie two copies of MatterControl running with each of the machine independently, but I am only running one printer at a time now).

As a side note, MatterControl is configured on my machine to pop up separate window panes instead of all in a block (like what is shown in the documentation).  At least once I had a pop-up window under another one, and it is possible, but unlikely, that I had an eprom settings window under the main window when I switched between printers.

larsbrubaker, 2016-05-26:
We have dramatically improved this and we also added import and export.

Fixed!

closed 2016-05-26 by larsbrubaker

ebo, 2016-05-27:
On May 26 2016 5:22 PM, Lars Brubaker wrote:
> We have dramatically improved this and we also added import and 
> export.
>
> Fixed!

Thanks, I will try to take a look at this, but it has been long enough 
that I will have to refresh everything.