MatterHackers/MatterControl #435: Temperature is never set when exporting gcode

ex-nerd, 2015-10-31:
I've been playing with the latest Mac MatterControl (1.4.0.5068) and just noticed that it doesn't actually seem to set the temperature (at least, not if you're just doing a straight gcode export -- MC isn't plugged directly into my Kossel Pro).

In reading through the gcode, the only temperature commands I see (M104, M109, M140) are the ones from my own manually-configured start/end gcode. These values differ significantly from the values I set for my filament, so it's pretty easy to tell.

cross-posted here: https://www.matterhackers.com/community/forum/topic/ag9zfm1hdHRlcmhhY2tlcnNyEgsSBVRvcGljGICAgKbwybcKDA

whosawhatsis, 2015-11-01:
It's standard practice to not set temperatures automatically when they are already set in your start code. You have to use replacement variables to retrieve the values from your settings when you set temperatures this way. There's a list of supported replacement variables here: https://github.com/MatterHackers/MatterControl/issues/51

ex-nerd, 2015-11-01:
Ah, interesting.  I guess I'm more accustomed to slicers with layer-specific temperature control so I've never run into this.  Seems odd to me, though.  I set the temperature high to squirt/wipe so was relying on mattercontrol to set things back to a sane level for the actual print.  I'll check out those variables and update my pre-print code accordingly (at least until you guys get around to supporting separate temps for different layers).

ex-nerd, 2016-05-14:
Confirmed that the custom variables work (though it'd still be nice to have different settings for multiple layers).  List of other variables is available on the wiki here: http://wiki.mattercontrol.com/Custom_G-Code_Variables

closed 2016-05-14 by ex-nerd