MatterHackers/MatterControl #436: Feature Request: first-layer filament temperature variation

ex-nerd, 2015-10-31:
Assuming #435 is fixed, it would be nice to have separate temperature settings for the first layer of filament.

rytz, 2016-12-27:
@larsbrubaker bumping this FR due to recent discussions about writing this feature.

Lodran, 2016-12-27:
Just to bump this a bit more, the specific use case I have for this is for Avante's FilaOne™ engineering grade filament.  The print surface used with FilaOne™ is a plastic sheet, and a lower temperature is required when extruding the first layer in order to avoid welding the print to the print sheet.  Other layers should be extruded at a higher temperature in order to promote welding between layers.

DanielJoyce, 2017-02-23:
I need this to print PLA reliably without it pulling away.

coosadog, 2017-11-03:
this is really a problem for me.  I've been trying to work around it with not much success.  I could do this with slic3r but that software is lacking.  I love matterslicer except for this.  I guess I could edit the gcode on every print but what a pain!!

ex-nerd, 2017-11-03:
@coosadog I've started using the Prusa Slic3r lately.  It's pretty good so far.

coosadog, 2017-11-03:
I liked slic3r pretty well but pronterface would get a lot of errors from the gcode.  Maybe I will try slic3r with mattercontrol.  It would seem that would be a nice mod to mattercontrol/matterslicer, though.

larsbrubaker, 2017-11-03:
You can achive this currently by adding a write filter that looks for "; LAYER 1:"

Here is a screen shot.

[image]

And the expected output:

G1 F1200 X103.853 Y60.861 E65.7903
; Layer Change GCode
M104 S210
G1 F3600 E63.2903

coosadog, 2017-11-03:
Wowwwww!!!  Great!  I didn't know this.  This is great news, thank you!!  I need to read more of the documentation, I think.  I'm really bad about that!  And I speed read a lot too...

coosadog, 2017-11-03:
That will be a great feature for MatterControl!  I'll use the filter until it comes out.  Thanks, again!

coosadog, 2017-11-08:
> We are working on tools that will allow varying the temp (and other
> settings) throughout the print. They will be visual and integrated into the
> 3D view, but for now this should get you what you need. :)

Lars?  I was looking on github and there has been no activity for at least
8 months....

larsbrubaker, 2017-11-09:
We have been doing our current work in the design_tools branch. There are tons of new features and improvements (including 64 bit and custom support).

[image]

Your can look at a dev build [here](https://mattercontrol.appspot.com/downloads/development/ag9zfm1hdHRlcmNvbnRyb2xyQQsSB1Byb2plY3QYgICA0KOn2goMCxINUHVibGljUmVsZWFzZRiAgIDAvYWYCgwLEgZVcGxvYWQYgICAwP3_hwoM) if you want.

coosadog, 2017-11-09:
OK, good, thanks!  Noooo, I don't do .exe's here.  I quit that habit 10
years ago!  I can't wait to get the new release!!

larsbrubaker, 2017-11-09:
Actually the installer is signed by matterhackers. Same as our normal releases.

coosadog, 2017-11-09:
Hey Lars.  I don't know what you mean by 'signed by matterhackers'.  How
can I get to the developer release, or can I?  I can build it here.  I'm a
retired software developer....  Been doing it my whole life!

coosadog, 2017-11-09:
hahaha!!!  OK, I know what you meant, now.  Sorry...

larsbrubaker, 2017-11-09:
:)

coosadog, 2017-11-09:
I haven't messed with C# or Mono in a very long time.  And then, only to
fix a bug or something in someone else's code.  I would never write
something from scratch in it.  I forgot how it works, is all.  I'll crank
it up in a little while, after this print terminates.  I'll give you
feedback, after that.

coosadog, 2017-12-12:
I finally figured out why my system was locking up.  It wasn't mono at all.  I wrote an opengl c++ program and I could make it do it.  I narrowed it down to my Nvidia driver.  I upgraded it and now everything seems to be stable.  You might want to make a note of that.