MatterHackers/MatterControl #456: Unable to Build with Fedora

HarlemSquirrel, 2015-11-15:
I am trying to build with Fedora but run into the following errors. I seem to be missing some files but do not see them in the Git repository and I tried deleting all my files and cloning again but keep getting stuck here. 
```
$ nuget restore MatterControl/MatterControl.slnWARNING: Project file /home/hs/MatterDevelop/MatterControl/../MatterControlAuth/MatterControlAuth/MatterControlAuth.csproj cannot be found.
WARNING: Project file /home/hs/MatterDevelop/MatterControl/../MatterControlPictureCreator/PictureCreator/PictureCreator.csproj cannot be found.
WARNING: Project file /home/hs/MatterDevelop/MatterControl/../CloudServicesPlugin/CloudServices/CloudServices.csproj cannot be found.
WARNING: Project file /home/hs/MatterDevelop/MatterControl/../CloudServicesPlugin/PrintNotifications/PrintNotifications.csproj cannot be found.
WARNING: Project file /home/hs/MatterDevelop/MatterControl/../X3GDriver/X3GDriver/X3GDriver.csproj cannot be found.
WARNING: Project file /home/hs/MatterDevelop/MatterControl/../CloudServicesPlugin/Mono.Nat/Mono.Nat.csproj cannot be found.
WARNING: Project file /home/hs/MatterDevelop/MatterControl/../CloudServicesPlugin/CloudServices.Tests/CloudServices.Tests.csproj cannot be found.
All packages listed in packages.config are already installed.
```

And then when I try to build anyway...
```
$ xbuild MatterControl/MatterControl.slnXBuild Engine Version 12.0
Mono, Version 4.0.5.0
Copyright (C) 2005-2013 Various Mono authors

Build started 11/15/2015 8:24:04 AM.
__________________________________________________
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/MatterControlAuth/MatterControlAuth/MatterControlAuth.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/MatterControlPictureCreator/PictureCreator/PictureCreator.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/CloudServicesPlugin/CloudServices/CloudServices.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/CloudServicesPlugin/PrintNotifications/PrintNotifications.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/X3GDriver/X3GDriver/X3GDriver.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/CloudServicesPlugin/Mono.Nat/Mono.Nat.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/CloudServicesPlugin/CloudServices.Tests/CloudServices.Tests.csproj referenced in the solution file, not found. Ignoring.
Project "/home/hs/MatterDevelop/MatterControl/MatterControl.sln" (default target(s)):
	Target ValidateSolutionConfiguration:
		Building solution configuration "Debug|Mixed Platforms".
	Target Build:
		Project "/home/hs/MatterDevelop/MatterControl/Submodules/agg-sharp/Tesselate/Tesselate.csproj" (default target(s)):
			Target PrepareForBuild:
				Configuration: Debug Platform: AnyCPU
			Target GenerateSatelliteAssemblies:
			No input files were specified for target GenerateSatelliteAssemblies, skipping.
			Target GenerateTargetFrameworkMonikerAttribute:
			Skipping target "GenerateTargetFrameworkMonikerAttribute" because its outputs are up-to-date.
			Target CoreCompile:
			Skipping target "CoreCompile" because its outputs are up-to-date.
		Done building project "/home/hs/MatterDevelop/MatterControl/Submodules/agg-sharp/Tesselate/Tesselate.csproj".
		Project "/home/hs/MatterDevelop/MatterControl/Submodules/agg-sharp/VectorMath/VectorMath.csproj" (default target(s)):
			Target PrepareForBuild:
				Configuration: Debug Platform: AnyCPU
			Target GenerateSatelliteAssemblies:
			No input files were specified for target GenerateSatelliteAssemblies, skipping.
			Target GenerateTargetFrameworkMonikerAttribute:
			Skipping target "GenerateTargetFrameworkMonikerAttribute" because its outputs are up-to-date.
			Target CoreCompile:
			Skipping target "CoreCompile" because its outputs are up-to-date.
		Done building project "/home/hs/MatterDevelop/MatterControl/Submodules/agg-sharp/VectorMath/VectorMath.csproj".
		Project "/home/hs/MatterDevelop/MatterControl/Submodules/agg-sharp/clipper_library/clipper_library.csproj" (default target(s)):
			Target PrepareForBuild:
				Configuration: Debug Platform: AnyCPU
			Target GenerateSatelliteAssemblies:
			No input files were specified for target GenerateSatelliteAssemblies, skipping.
			Target GenerateTargetFrameworkMonikerAttribute:
			Skipping target "GenerateTargetFrameworkMonikerAttribute" because its outputs are up-to-date.
			Target CoreCompile:
			Skipping target "CoreCompile" because its outputs are up-to-date.
		Done building project "/home/hs/MatterDevelop/MatterControl/Submodules/agg-sharp/clipper_library/clipper_library.csproj".
		Project "/home/hs/MatterDevelop/MatterControl/Submodules/MatterSlice/MatterSlice.csproj" (default target(s)):
			Target PrepareForBuild:
				Configuration: Debug Platform: AnyCPU
			Target GenerateSatelliteAssemblies:
			No input files were specified for target GenerateSatelliteAssemblies, skipping.
			Target GenerateTargetFrameworkMonikerAttribute:
			Skipping target "GenerateTargetFrameworkMonikerAttribute" because its outputs are up-to-date.
			Target CoreCompile:
				Tool /usr/lib/mono/4.5/mcs.exe execution started with arguments: /noconfig /debug:full /debug+ /optimize- /out:obj/Debug/MatterSlice.exe Comb.cs GCodePathConfig.cs GCodePlanner.cs Properties/AssemblyInfo.cs layerPart.cs sliceDataStorage.cs Slicer/SlicerLayer.cs Slicer/Slicer.cs NewSupport.cs utils/clipper.cs utils/floatpoint.cs main.cs Settings.cs timeEstimate.cs utils/IntpointHelper.cs utils/SplitCommandLine.cs utils/StringHelper.cs fffProcessor.cs gcodeExport.cs optimizedModel.cs SimpleModel.cs AvoidCrossingPerimeters.cs utils/PolygonHelper.cs PathOrderOptimizer.cs support.cs bridge.cs logoutput.cs infill.cs TopsAndBottoms.cs skirt.cs raft.cs multiVolumes.cs inset.cs /target:exe /unsafe- /define:"DEBUG;TRACE" /nostdlib /platform:x86 /reference:/usr/lib/mono/4.5/System.dll /reference:/usr/lib/mono/4.5/System.Core.dll /reference:/usr/lib/mono/4.5/mscorlib.dll /warn:4
[... 57 lines cut ...]
Errors:

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->
/home/hs/MatterDevelop/MatterControl/Submodules/MatterSlice/MatterSlice.csproj (default targets) ->
/usr/lib/mono/4.5/Microsoft.CSharp.targets (CoreCompile target) ->

	utils/IntpointHelper.cs(155,32): error CS0103: The name `PI' does not exist in the current context

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->

	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlAuth/MatterControlAuth/MatterControlAuth.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlPictureCreator/PictureCreator/PictureCreator.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices/CloudServices.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/PrintNotifications/PrintNotifications.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/X3GDriver/X3GDriver/X3GDriver.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/Mono.Nat/Mono.Nat.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices.Tests/CloudServices.Tests.csproj'

	 13 Warning(s)
	 8 Error(s)

Time Elapsed 00:00:01.6292240
```

unlimitedbacon, 2015-11-20:
The missing files are references to closed source plugins. It will build without them. The real error is
```
utils/IntpointHelper.cs(155,32): error CS0103: The name `PI' does not exist in the current context
```
You can work around tha by changing line 155 in `Submodules/MatterSlice/utils/IntpointHelper.cs` from
```
rotation = rotation / 180 * PI;
```
to
```
rotation = rotation / 180 * Math.PI;
```

HarlemSquirrel, 2015-11-20:
Thanks for the tip! I won't have access to this machine for another week but I am definitely going to try this then.

HarlemSquirrel, 2015-11-27:
OK, so the Math.PI change got me further I think. But I still have some errors. 

```
CSC: error CS2001: Source file `PrinterCommunication/RepRapReadWriteCallbacks.cs' could not be found
			Task "Csc" execution -- FAILED
			Done building target "CoreCompile" in project "/home/hs/MatterDevelop/MatterControl/MatterControl.csproj".-- FAILED
		Done building project "/home/hs/MatterDevelop/MatterControl/MatterControl.csproj".-- FAILED
	Task "MSBuild" execution -- FAILED
	Done building target "Build" in project "/home/hs/MatterDevelop/MatterControl/MatterControl.sln".-- FAILED
Done building project "/home/hs/MatterDevelop/MatterControl/MatterControl.sln".-- FAILED

Build FAILED.

Warnings:

/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/MatterControlAuth/MatterControlAuth/MatterControlAuth.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/MatterControlPictureCreator/PictureCreator/PictureCreator.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/CloudServicesPlugin/CloudServices/CloudServices.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/CloudServicesPlugin/PrintNotifications/PrintNotifications.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/X3GDriver/X3GDriver/X3GDriver.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/CloudServicesPlugin/Mono.Nat/Mono.Nat.csproj referenced in the solution file, not found. Ignoring.
/home/hs/MatterDevelop/MatterControl/MatterControl.sln:  warning : Project file /home/hs/MatterDevelop/CloudServicesPlugin/CloudServices.Tests/CloudServices.Tests.csproj referenced in the solution file, not found. Ignoring.

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->
/home/hs/MatterDevelop/MatterControl/Submodules/MatterSlice/MatterSlice.csproj (default targets) ->
/usr/lib/mono/4.5/Microsoft.CSharp.targets (CoreCompile target) ->

	Slicer/SlicerLayer.cs(137,5): warning CS0162: Unreachable code detected
	Slicer/SlicerLayer.cs(613,10): warning CS0219: The variable `a' is assigned but its value is never used
	NewSupport.cs(58,17): warning CS0219: The variable `selfSupportDistance' is assigned but its value is never used
	Settings.cs(546,17): warning CS0219: The variable `lines' is assigned but its value is never used
	Settings.cs(566,13): warning CS0219: The variable `value' is assigned but its value is never used
	fffProcessor.cs(439,24): warning CS0219: The variable `newSupport' is assigned but its value is never used
	fffProcessor.cs(785,6): warning CS0162: Unreachable code detected
	gcodeExport.cs(107,12): warning CS0219: The variable `numberString' is assigned but its value is never used
	PathOrderOptimizer.cs(84,14): warning CS0219: The variable `rotatedPrev' is assigned but its value is never used
	GCodePlanner.cs(57,18): warning CS0414: The private field `MatterHackers.MatterSlice.GCodePlanner.totalPrintTime' is assigned but its value is never used
	Settings.cs(33,15): warning CS0414: The private field `MatterHackers.MatterSlice._ConfigSettingIndex.ptr' is assigned but its value is never used
	infill.cs(293,27): warning CS0414: The private field `MatterHackers.MatterSlice.Infill.hexOffset' is assigned but its value is never used

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->
/home/hs/MatterDevelop/MatterControl/Submodules/MatterSlice/Tests/MatterSlice.Tests/MatterSlice.Tests.csproj (default targets) ->
/usr/lib/mono/xbuild/12.0/bin/Microsoft.CSharp.targets (CoreCompile target) ->

	MatterSlice/GCodePlannerTests.cs(153,14): warning CS0219: The variable `correctedPath' is assigned but its value is never used
	MatterSlice/SlicerLayerTests.cs(183,45): warning CS0219: The variable `segmentsList' is assigned but its value is never used

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->
/home/hs/MatterDevelop/MatterControl/MatterControl.csproj (default targets) ->
/usr/lib/mono/4.5/Microsoft.Common.targets (ResolveAssemblyReferences target) ->

	/usr/lib/mono/4.5/Microsoft.Common.targets:  warning : Reference 'PresentationFramework' not resolved

Errors:

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->

	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlAuth/MatterControlAuth/MatterControlAuth.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlPictureCreator/PictureCreator/PictureCreator.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices/CloudServices.csproj'
[... cut ...]
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices.Tests/CloudServices.Tests.csproj'

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->
/home/hs/MatterDevelop/MatterControl/MatterControl.csproj (default targets) ->
/usr/lib/mono/4.5/Microsoft.CSharp.targets (CoreCompile target) ->

	CSC: error CS2001: Source file `PrinterCommunication/RepRapReadWriteCallbacks.cs' could not be found

	 22 Warning(s)
	 8 Error(s)

Time Elapsed 00:00:08.5659210
```

mrunion, 2015-11-29:
The file "RepRapReadWriteCallBacks.cs" is the new name for "RepRapReadWriteCallbacks.cs" (notice the case diference?). When building on a non-case-sensitive OS, this stuff is hard to catch, but the issue is the case.

mrunion, 2015-11-29:
OK, they mispelled "GetFilamentDiameter" in the agg-sharp/examples/GCodeVisualizer code on my copy of the code that was pulled. See if you have the wrong spelling (the function was called "GetFilamentDiamter") on line 85 in the source for agg-sharp/examples/GCodeVisualizer/GCodeFile.cs

mrunion, 2015-11-29:
And finally, in the Tets folder, PrinterCallBackTests.cs is the existing file, but PrinterCallbackTests.cs is the file the code tries to compile. After fixing that issue, I finally got this version to compile.

FWIW: Arch Linux, Knockoff Prusa I3 single extruder, Prusa i3 with dual extruder being shipped, heated beds, one acrylic frame and the new printer has a steel frame.

HarlemSquirrel, 2015-11-30:
Nice finds! I will try to compile this when I get back to this machine. Did you sumbit a pull request for these fixes?

mrunion, 2015-12-09:
No, I didn't do a pull request. I have been out of town because of a medical issue with a family member. I will revisit this soon, as time permits and try to make a pull request.

jlewin, 2015-12-17:
Linux builds should be back on track and are once again passing on Ubuntu. Please advise if you're still running into compile issues

HarlemSquirrel, 2015-12-20:
I am still unable to build on Fedora 23. I attached the full output when attempting to build. 
These are the errors:
```
/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->

	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlAuth/MatterControlAuth/MatterControlAuth.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlPictureCreator/PictureCreator/PictureCreator.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices/CloudServices.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/PrintNotifications/PrintNotifications.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/X3GDriver/X3GDriver/X3GDriver.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/Mono.Nat/Mono.Nat.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices.Tests/CloudServices.Tests.csproj'

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->
/home/hs/MatterDevelop/MatterControl/MatterControl.csproj (default targets) ->
/usr/lib/mono/4.5/Microsoft.CSharp.targets (CoreCompile target) ->

	CSC: error CS2001: Source file `PrinterCommunication/RepRapReadWriteCallbacks.cs' could not be found
```
[MatterControl_build_fail_2015-12-20.txt](https://github.com/MatterHackers/MatterControl/files/67818/MatterControl_build_fail_2015-12-20.txt)

mrunion, 2015-12-20:
Check the case of that RepRapReadWriteCallbacks file. I bet you'll find its RepRapReadWriteCallBacks.cs on the file system. Rename it and see if it compiles.

Thanks,
Matt Runion
mrunion@yahoo.com

HarlemSquirrel, 2015-12-20:
Nope, now I get 
```
/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->

	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlAuth/MatterControlAuth/MatterControlAuth.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlPictureCreator/PictureCreator/PictureCreator.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices/CloudServices.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/PrintNotifications/PrintNotifications.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/X3GDriver/X3GDriver/X3GDriver.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/Mono.Nat/Mono.Nat.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices.Tests/CloudServices.Tests.csproj'

/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->
/home/hs/MatterDevelop/MatterControl/MatterControl.csproj (default targets) ->
/usr/lib/mono/4.5/Microsoft.CSharp.targets (CoreCompile target) ->

	CSC: error CS2001: Source file `PrinterCommunication/RepRapReadWriteCallbacks.cs' could not be found
```

HarlemSquirrel, 2015-12-20:
Nevermind, I renamed it wrong! Now I have `CSC: error CS2001: Source file `MatterControl/PrinterCallbackTests.cs' could not be found`

HarlemSquirrel, 2015-12-20:
OK, so now I have...
```
/home/hs/MatterDevelop/MatterControl/MatterControl.sln (default targets) ->
(Build target) ->

	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlAuth/MatterControlAuth/MatterControlAuth.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/MatterControlPictureCreator/PictureCreator/PictureCreator.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices/CloudServices.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/PrintNotifications/PrintNotifications.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/X3GDriver/X3GDriver/X3GDriver.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/Mono.Nat/Mono.Nat.csproj'
	/home/hs/MatterDevelop/MatterControl/MatterControl.sln: error : Could not find the project file '/home/hs/MatterDevelop/CloudServicesPlugin/CloudServices.Tests/CloudServices.Tests.csproj'
```

jlewin, 2015-12-20:
The missing project build errors are to be expected and can be ignored. In the near future those projects will be removed from MatterControl.sln to make this clearer and resolve the errors. Assuming MatterControl.exe exists in your output directory, does it run successfully?

Also, you certainly shouldn't have to rename files to build the solution. The RepRapReadWriteCallBacks casing issue was addressed back on Dec. 3rd in the following:

https://github.com/MatterHackers/MatterControl/commit/c1e144d467d9ed26eb7edc03683e0de8e2d1ab94

Are you on *master* and all synced up? Are your submodules synced and/or does git status report "up-to-date" with nothing in the "modified" state? I see ten successful builds on our Ubuntu servers using xbuild (which is case sensitive) in the last couple of days so I'm a bit confused and definitely want to understand what you're seeing.

Thanks,

HarlemSquirrel, 2015-12-21:
Hmm, OK. I won't be back on this machine until next weekend but I will try it again. Thank you for your help!

HarlemSquirrel, 2015-12-27:
I am still unable to build on the master branch or 1.4. I get a lot of files in bin/release but no MatterControl.exe
```
$ ls bin/Release/
AGG.dll                                      MatterHackers.OpenGL.UI.dll
agg_platform_win32.dll                       MatterHackers.OpenGL.UI.dll.mdb
C5.dll                                       MatterHackers.PolygonMesh.dll
clipper_library.dll                          MatterHackers.PolygonMesh.dll.mdb
clipper_library.dll.mdb                      MatterHackers.RenderOpenGl.dll
Community.CsharpSqlite.dll                   MatterHackers.SerialPortCommunication.dll
GCodeVisualizer.exe                          MatterHackers.SerialPortCommunication.dll.mdb
GuiAutomation.dll                            MatterHackers.VectorMath.dll
GuiAutomation.dll.mdb                        MatterHackers.VectorMath.dll.mdb
InfInstaller.exe                             MatterSlice.exe
InfInstaller.exe.mdb                         MatterSlice.exe.mdb
Launcher.exe                                 Mindscape.Raygun4Net4.dll
Launcher.exe.mdb                             Mindscape.Raygun4Net.dll
MatterHackers.Agg.ImageProcessing.dll        Newtonsoft.Json.dll
MatterHackers.Agg.ImageProcessing.dll.mdb    OpenTK.dll
MatterHackers.Agg.UI.dll                     OpenTK.GLControl.dll
MatterHackers.Agg.UI.dll.mdb                 PdfSharp.dll
MatterHackers.Csg.dll                        PlatfromAbstract.dll
MatterHackers.Csg.dll.mdb                    Raygun.Diagnostics.dll
MatterHackers.DataConverters3D.dll           RayTracer.dll
MatterHackers.DataConverters3D.dll.mdb       RayTracer.dll.mdb
MatterHackers.Localizations.dll              STLViewer.exe
MatterHackers.MarchingSquares.dll            Tesselate.dll
MatterHackers.MatterControl.Pulgins.dll      Tesselate.dll.mdb
MatterHackers.MatterControl.Pulgins.dll.mdb  WindowsFileDialogs.dll
```

jlewin, 2015-12-27:
It's been a very long time since I've touched Fedora or RedHat, but I setup a Fedora 23 x64 vm on DigitalOcean over a slow satellite connection and was able to build MatterControl. The basic sequence was something along the lines of:

```
yum install git
yum install mono-devel
yum install nuget
mozroots --import --sync

mono --version (must be v4.0.5 or greater)

git clone --recursive https://github.com/MatterHackers/MatterControl.git
cd MatterControl
nuget restore MatterControl.sln

xbuild MatterControl.sln
```

This produced 7 build errors due to the missing external projects but the MatterControl.exe output file was generated as expected.

In addition, I went through the process a second time on second clean VM sticking only to the commands depicted above to ensure I didn't missing any required steps. As with our Ubuntu build servers I can only confirm the build process runs successfully and more work would be needed to test the exe in a graphical environment. 

A few of things that seem likely to break a local build would be:

 - Submodules out of sync with the main repository
 - Older versions on mono that might not have C# 6 build compatibility
 - Nuget restore issues

Feel free to post your latest build logs for review

Thanks,

HarlemSquirrel, 2015-12-27:
OK, so I cloned to a new directory following these instructions and now I have MatterControl.exe in `bin/Debug/` but she won't run :(
```
$ mono bin/Debug/MatterControl.exe
[ERROR] FATAL UNHANDLED EXCEPTION: System.IO.DirectoryNotFoundException: Could not find a part of the path "/home/hs/builds/MatterControl/bin/Debug/StaticData/OEMSettings/Settings.json".
  at System.IO.FileStream..ctor (System.String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, Boolean anonymous, FileOptions options) <0x7f6028ba0a30 + 0x001fb> in <filename unknown>:0 
  at System.IO.FileStream..ctor (System.String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, System.String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost) <0x7f6028ba0980 + 0x0004d> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream:.ctor (string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int,System.IO.FileOptions,string,bool,bool,bool)
  at System.IO.StreamReader..ctor (System.String path, System.Text.Encoding encoding, Boolean detectEncodingFromByteOrderMarks, Int32 bufferSize, Boolean checkHost) <0x7f6028a0f7d0 + 0x000e7> in <filename unknown>:0 
  at System.IO.StreamReader..ctor (System.String path, System.Text.Encoding encoding, Boolean detectEncodingFromByteOrderMarks, Int32 bufferSize) <0x7f6028a0f790 + 0x00030> in <filename unknown>:0 
  at System.IO.StreamReader..ctor (System.String path, Boolean detectEncodingFromByteOrderMarks) <0x7f6028a0f6c0 + 0x0003b> in <filename unknown>:0 
  at System.IO.StreamReader..ctor (System.String path) <0x7f6028a0f6a0 + 0x00016> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.IO.StreamReader:.ctor (string)
  at System.IO.File.ReadAllText (System.String path) <0x7f6028b9ed40 + 0x00036> in <filename unknown>:0 
  at MatterHackers.Agg.FileSystemStaticData.ReadAllText (System.String path) <0x40951880 + 0x00023> in <filename unknown>:0 
  at MatterHackers.MatterControl.SettingsManagement.OemSettings.get_Instance () <0x409517e0 + 0x00059> in <filename unknown>:0 
  at MatterHackers.MatterControl.ActiveTheme..ctor () <0x4094eca0 + 0x0011b> in <filename unknown>:0 
  at MatterHackers.MatterControl.ActiveTheme.get_Instance () <0x4094ec10 + 0x0003b> in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication..ctor (Double width, Double height) <0x40949c20 + 0x00fe3> in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.CreateInstance () <0x40914200 + 0x001d3> in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.get_Instance () <0x40914190 + 0x0001b> in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () <0x408d4090 + 0x001f7> in <filename unknown>:0
```

mrunion, 2015-12-29:
I have been and still am away from my machine. But I read the error message(s) above and I wonder if the path it mentions is there, and if everything is in the same case:

Could not find a part of the path "/home/hs/builds/MatterControl/bin/Debug/StaticData/OEMSettings/Settings.json".

Is that file present? Same case?

johnlewin, 2015-12-30:
MatterControl has some unique behavior around the StaticData directory that looks for a local copy in the working directory or assumes you're running from bin/Debug or bin/Release and attempts to load the repo copy from the parent path. Try changing to ```bin/Debug``` before running ```mono MatterControl.exe```

HarlemSquirrel, 2015-12-30:
johniewin, you were right! This worked like a charm!

closed 2015-12-30 by HarlemSquirrel

johnlewin, 2015-12-30:
Great to hear you got it working! Thanks for hanging in there and keeping at it and know that we're working to improve the build process. Hopefully in the new year we'll have an update that revamps our projects to eliminate build errors and reduce warnings.