MatterHackers/MatterControl #474: Replicator 2 Settings

HarlemSquirrel, 2015-12-11:
I tried the configuration file on the wiki and had a lot of problems. The configuration settings in MatterControl 1.4 are better, but I incorporated a few things from the file on the wiki and a few things I noticed in the MakerWare settings. 

My new configuration is much better, but not without issues. 
 - Layer heights are sometimes inconsistent.
 - Canceling on the printer causes the bed to drop but the extruder to continue to print!
 - Canceling too early can cause the print bed to try to move past it's lowest point and 'grind' until I turn the printer off.

Here is my slice file 
[MB_Rep2.slice.txt](https://github.com/MatterHackers/MatterControl/files/59501/MB_Rep2.slice.txt)

Here is my cancel gcode:
`cancel_gcode = M104 S0 ;turn off temperature \nM140 S0 ;turn off bed temperature\nG0 Z155 ;Send Z axis to bottom of machine\nG28 X0 Y0\nM1201 ;Build End Notification MakerBot\n`

HarlemSquirrel, 2015-12-14:
I made some more adjustments to my configurations and I'm having more success! I was having trouble with stringing and prints falling apart on small areas. I decreased the PLA extruder temp to 220C, increased the min layer print time to 4 sec, decreased the minimum print speed to 10 mm/s, and a few other things. 
Take a look at this small ice cycle print! http://www.thingiverse.com/make:177510

[MB_Rep2 v2.slice.txt](https://github.com/MatterHackers/MatterControl/files/61672/MB_Rep2.v2.slice.txt)

larsbrubaker, 2016-05-26:
Check if we want to adopt these changes, @mmoening.