MatterHackers/MatterControl #493: Crash

jschall, 2015-12-29:
'''
Unhandled Exception:
System.NullReferenceException: Object reference not set to an instance of an object
  at System.Windows.Forms.XEventQueue+PaintQueue.Dequeue () [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUIX11.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr handle, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUI.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr hWnd, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.RunLoop (Boolean Modal, System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.Run (System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.Run (System.Windows.Forms.Form mainForm) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WidgetForWindowsFormsAbstract.Run () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WidgetForWindowsFormsOpenGL.Run () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.SystemWindowCreator_WindowsForms.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.get_Instance () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () [0x00000] in <filename unknown>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object
  at System.Windows.Forms.XEventQueue+PaintQueue.Dequeue () [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUIX11.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr handle, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUI.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr hWnd, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.RunLoop (Boolean Modal, System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.Run (System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.Run (System.Windows.Forms.Form mainForm) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WidgetForWindowsFormsAbstract.Run () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WidgetForWindowsFormsOpenGL.Run () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.SystemWindowCreator_WindowsForms.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.get_Instance () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () [0x00000] in <filename unknown>:0
'''
Happened between hitting "Print" and anything happening on the printer. Can't reproduce regularly with the same file.

jschall, 2015-12-29:
Ubuntu 15.10 MatterControl 1.4.0.5485

neurobomber, 2016-01-05:
I'm getting this problem as well. I'm using Lubuntu 15.10. Here's my own output:

```
Unhandled Exception:
System.NullReferenceException: Object reference not set to an instance of an object
  at System.Windows.Forms.XEventQueue+PaintQueue.Dequeue () [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUIX11.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr handle, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUI.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr hWnd, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.RunLoop (Boolean Modal, System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.Run (System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.Run (System.Windows.Forms.Form mainForm) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WidgetForWindowsFormsAbstract.Run () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WidgetForWindowsFormsOpenGL.Run () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.SystemWindowCreator_WindowsForms.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.get_Instance () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () [0x00000] in <filename unknown>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object
  at System.Windows.Forms.XEventQueue+PaintQueue.Dequeue () [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUIX11.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr handle, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUI.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr hWnd, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.RunLoop (Boolean Modal, System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.Run (System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.Run (System.Windows.Forms.Form mainForm) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WidgetForWindowsFormsAbstract.Run () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.WidgetForWindowsFormsOpenGL.Run () [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.SystemWindowCreator_WindowsForms.ShowSystemWindow (MatterHackers.Agg.UI.SystemWindow systemWindow) [0x00000] in <filename unknown>:0 
  at MatterHackers.Agg.UI.SystemWindow.ShowAsSystemWindow () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.get_Instance () [0x00000] in <filename unknown>:0 
  at MatterHackers.MatterControl.MatterControlApplication.Main () [0x00000] in <filename unknown>:0 
```

neurobomber, 2016-01-05:
I seem to have found a fix, at least for this current print that was consistently coming to this error at a specific layer.

Go to Settings -> General -> Repair and check both of the boxes in that dialog. I'll report back if I ever see this again.

neurobomber, 2016-01-29:
Never mind, I'm still encountering this issue on random prints and random times. Not sure what's causing it but I've encountered it on multiple machines with Ubuntu installed.

unlimitedbacon, 2016-02-26:
This is a duplicate of [this issue](https://github.com/MatterHackers/MatterControl/issues/378).

xf3qc, 2016-03-18:
Same issue here.

Found similar problem here
[https://github.com/MatterHackers/MatterControl/issues/493](url)

 with possible fix:

` answered 2013-02-04 19:03 kpe

I had the same problem (.net, Linux, Mono). My program has a provider-subsrciber pattern. The provider is on another thread. The main form is subscribed to the provider. When it is fired:

    private void RefreshLabels(ParameterMap pm)
    {
        string StateValueKey = string.Empty;
        string svkValue = string.Empty;

        if (InvokeRequired)
        {
            BeginInvoke(new RefreshItemsDelegate(RefreshLabels), new object[] { pm });
        }
        else
        {

..... it comes across from the other thread and does its job.

Another subscriber fired an event on the main form also. I ignored this event and the problem DISAPPEARED.`

`dFromPrinter thread created.
Time To Load Seconds: 1.85

Unhandled Exception:
System.NullReferenceException: Object reference not set to an instance of an object
  at System.Windows.Forms.XEventQueue+PaintQueue.Dequeue () [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUIX11.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr handle, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.XplatUI.GetMessage (System.Object queue_id, System.Windows.Forms.MSG& msg, IntPtr hWnd, Int32 wFilterMin, Int32 wFilterMax) [0x00000] in <filename unknown>:0 
  at System.Windows.Forms.Application.RunLoop (Boolean Modal, System.Windows.Forms.ApplicationContext context) [0x00000] in <filename unknown>:0 `

xf3qc, 2016-03-19:
ok, so it seems that changing 3D view - to - Layer view stops crashing
Perhaps it fails to draw something for 3d render printing preview screen and crashes mattercontrol

xf3qc, 2016-04-16:
nope. still crashing.
No follow up from anyone..
:(

Hi-Angel, 2016-06-30:
I guess it's a bug in the Mono itself. I think so because at my work I'm also making a GUI in C#, most of the testing is done on GNU/Linux — and the app occasionally crashes with a similar backtrace; and the only mine code in there is the Main function, then it goes to `Application.Run()` *(as it is in the backtrace here)*, and goes to execute Mono Forms internals.

I can't reproduce, tho, i.e. the crash is occasional. The only common thing in *my* crash is that it always happen when I'm switching the desktop. Also, not sure if that matters, but I do use a tiling WM.

A workaround, I think, is to use a more advanced Tookits, like Qt or GTK. In the end, WinForms didn't have much of a usual functionality. It didn't implement even the primary selection, to begin with.

unlimitedbacon, 2016-06-30:
We have basically come to the same conclusion as well. It seems to be a bug in Mono's implementation of Windows.Forms, which Xamarin is no longer interested in supporting. We have never been able to isolate any steps to reproduce it either. Long term, the solution is to move to GTK, which would probably fix a number of other minor issues as well, but it would be quite an undertaking.

larsbrubaker, 2018-09-24:
Closing.
We have a new issue for porting linux to GTK and that will address this issue

closed 2018-09-24 by larsbrubaker

Hi-Angel, 2018-09-24:
@larsbrubaker could you please link the issue, typing `gtk` or `gtk3` in search bar of issue doesn't seem to get me anything relevant.