MatterHackers/MatterControl #534: "Unpause Resets XYZ" Makes Manual Levelling Impossible

Parmeisan, 2016-01-19:
I have been using and enjoying MatterControl for about a year now, so thank you for all your hard work. :)

I have a BIG problem with the update, though.  My Auto-Levelling never worked, but at least I could always pause the print and manually move the Z axis to correct it.  Now I cannot, and after fiddling for a couple of hours I am forced to conclude that MatterControl is useless to me after this update.  I can't seem to revert to the prior version either, so hopefully either this can be fixed, or I can find other software that works for me.  I like MatterControl, so please consider making this feature optional - or turning it off.

Thanks!

larsbrubaker, 2016-01-20:
@Parmeisan, thanks for the feedback. There is nothing that I know of in the new build that would cause this behavior. Specifically the slicer outputs absolute coordinates and adjusting the z height while  paused will not change the z position of future gcode. Is it possible you were only wanting this effect for a single layer? It is possible that some behavior changed there.

More importantly the next version will have a great fix for this. We have included the ability to adjust the z height while printing and it will stay adjusted through the entire print. This new version should be out by the end of this week in alpha.

Parmeisan, 2016-01-20:
That's a bit odd, because I would have sworn that it was acting that way before, but not now.  Also, if I compare 1.4.1 to 1.5.0, I see a comment that says "Making the pause an resume always reset the extrusion position and go back to the last xyz"... but I guess that's something that's being added to the alpha, not something that's currently in production?

Well, I'm familiar enough with users incorrectly reporting errors because they remember it wrong, just not usually from this end. :)  Let's assume then that I was mistaken.  Part of this may be related to my discovery late last night that the extrusion had more or less stopped - perhaps at the precise moment that I began to experiment with disabling Automatic Leveling.  At any rate, my previous efforts appear to include some faulty assumptions, so I am content to ignore them.

Two things, then:

1) That sounds like a great update and one that I will use often.  I'm not sure how to compile the code that can be downloaded from GitHub, though... I assume I will need to figure this out in order to try the alpha build?  Or would you mind sending me an .exe?  I can help to test, since I was planning on doing a lot of printing in the next week or two.

2) I have discovered a temporary workaround wherein I can _manually_ manually adjust the Z by rotating the cylinders on either side of the heat bed.  This is probably not great for the printer, but for now it works.  This morning I have started a print that appears to be working normally.

Thanks for your quick and helpful reply. :)

Parmeisan, 2016-01-20:
PS. Can we set the issue to "Pending" or anything like that?  I'm not sure that it's Closed, but it's hardly Open either.  Perhaps it could be labelled "Enhancement"; I see that other issues have been tagged as such, but I don't think I can do it.

larsbrubaker, 2016-01-20:
@Parmeisan [here](https://mattercontrol.appspot.com/downloads/development/ag9zfm1hdHRlcmNvbnRyb2xyOwsSB1Byb2plY3QY6gcMCxINUHVibGljUmVsZWFzZRiAgICY8MOJCgwLEgZVcGxvYWQYgICAmJqRmQoM) is a link to our current release candidate. If this thing can make it through test, this will go alpha later today. 

I've updated the status of this too. :)

Parmeisan, 2016-01-20:
Excellent!  My current print will be done in 1 or 2 hours.  I will use this for my next print, shortly after it's done.

Parmeisan, 2016-01-21:
All right... I have had no end of trouble this afternoon, but I am fairly confident that it's all to the blame of my printer and myself, not the software.

Regarding the software, it did tell me at first that I couldn't create the folder I wanted to install into.  That was just the default "C:/Program Files (x86)/MatterControl".  This might be because I had the parent folder open in Windows.  I closed it and tried again, and it worked.

The Z+ and Z- have been working well for me.  I have used the 0.1 and 1 values quite a bit, though I haven't had occasion to use X, Y, or any of the other increments.  I notice that the retract and extrude buttons are still greyed out, but I expect that's probably intentional, at least for now.  I have had no other problems with Matter Control.

I don't know if you're going to make other changes to the software -  or even if this reply comes too late and you've already pushed it out -  but I'm content.  Thanks so much for your help.

closed 2016-01-21 by Parmeisan

larsbrubaker, 2016-01-21:
Great to hear!