MatterHackers/MatterControl #545: Slic3r within SeeMeCNC version of MC not positioning correctly

jjjohnson4112, 2016-02-01:
When using Slic3r as the slicing engine, positioning STL in MC works fine, but when sliced it positions the part in the upper right quadrant of the build area. Photos attached. The temporary  work around is to adjust the print center to 100,100 when using Slic3r. 

[image]
[image]
[image]

jjjohnson4112, 2016-04-12:
Still broken? Any news on a fix?

larsbrubaker, 2016-04-12:
Hey JJ, we have not had a chance to dig into this yet. Hopefully we will soon.

jjjohnson4112, 2016-04-12:
Awesome. Thanks for the update.