MatterHackers/MatterControl #579: Feature Request: Upload/Download Settings or Include Settings in Model "Uploaded"

gwestwater, 2016-02-25:
I use MatterTouch on my Robo3D. I use Matter Control on my desktop to view / manage the models. I would love the ability to "Send" the settings to the MatterTouch or download them from the MatterTouch.

Even better would be to be able to send the settings with the model. For example that would let me set the speed on a specific print, or enable / disable raft etc... It would make repeating a good print in the future much more reliable

larsbrubaker, 2016-05-26:
The ability to share settings across devices is on our roadmap and is coming in a future release. We're also interested in the per model settings but have not yet scheduled this work.

larsbrubaker, 2016-05-26:
Thanks again, these issues have been added to our planned development: #809, #810

closed 2016-05-26 by larsbrubaker