MatterHackers/MatterControl #583: Temperature control oscillates on cancel + preheat request.

indrora, 2016-02-27:
I seem to hit this edge case a lot.

When canceling a print, then immediately requesting a preheat (e.g. within a 5...10 seconds of canceling) so that the hot end stays hot, MatterControl will ping back and forth between a target temperature of 0 and the preheat temperature, then stop accepting temperature updates. In that time, the serial queue is so full of temperature requests, any requests to home the axes of the printer are delayed and MatterControl tends to crash.

jlewin, 2016-05-12:
Thanks for the bug report. This issue was recently fixed in #653 and will be making its way into our next update.

johnlewin, 2016-05-12:
Fixed via 375321c8d53f81af5365ea4e43145eb

closed 2016-05-12 by johnlewin