MatterHackers/MatterControl #596: MatterControl 1.5 Requests

RogerRepRap, 2016-03-04:
New user coming from slic3r. Like Matter control, just a couple of nits.
1. Expand layer height box. If you use Prusa calculator with imperial lead screws you get 5 digit (x.xxxx) layer heights. Box is too small to see entry.
2. Open model dialog does not remember where last model was loaded from. If you are working a bunch of projects at one time with a bunch of parts this can get annoying.
3. I like the way that when you save your configuration it saves all settings. What I don't like is that they are not saved and looked for in a single location. Loading configuration dialog should be obvious and the open dialog should reflect this location. It looks to me like they are saved where the part resides.
4. I like the ability to combine infill every X layers. In my experience gives model higher shear strength and cuts print time.

Roger

RogerRepRap, 2016-03-04:
Oops... 2 is not a problem. Works as expected.

Renha, 2016-04-05:
I think these nits are better to be as separate issues [sorry my engrish]