MatterHackers/MatterControl #618: Support "Air Gap" is being ignored

jjjohnson4112, 2016-03-15:
The air gap setting for support material is being ignored. The support is being built all the way up to the model without any air gap at all. 

The attached image shows an exaggerated setting of a 10mm air gap ( I usually use .2-.3 but supports were seemingly impossible to remove cleanly, and on frail parts it can't be removed without destroying the part.)

[image]

jjjohnson4112, 2016-04-12:
Lars,

Support generation is still misbehaving and causing inconsistent and unexpected results. I have created a Short video here to show what is happening: https://youtu.be/rEcmscCGLSA

The gist is that MatterControl is forcing the user to use interface layers. If the user sets that to 0 the airgap is completely ignored for the print. Users are wanting to print without the interface layer(s)

If the user has interface layers, the air gap is applied, but has the order of items to print incorrect. It raises up to print the interface layer, then returns lower to print the in between layer(s) of the part. This is greatly exaggerated in the video since I have a 3mm air gap

larsbrubaker, 2016-05-07:
Patch 58f13d68ed9ada8122898c67a982f63d2805f856 in MatterSlice fixes this issue.

It was due to the interface layer count being set to 0. It now will work when 0 interface layers.

I don't normally recommend printing without interface layers, but I'm sure there are times that that is what is needed.

closed 2016-05-07 by larsbrubaker

larsbrubaker, 2016-05-07:
Thanks for the great report and the video. Very helpful.