MatterHackers/MatterControl #653: Temperature Freakout

jjjohnson4112, 2016-04-05:
Occasionally when using the manual controls within MatterControl there will be a software lockup after changing the temperature more than once. This causes the temperature readout to flash between the original set point and the updated temp in a neverending loop until the program is forced to close and restarted. This seems to be an issue since the 1.5 version.

larsbrubaker, 2016-04-05:
Thanks for the report. We will get this on the list of bugs.

johnoly99, 2016-04-11:
Lars

Here's a link to a screencapture we took during running macros.  It's pretty weird?  We've seen it both in the desktop when clicking/setting temps, and when running macros, but if i type line by line one at a time in the terminal, it doesn't do it at all?

https://youtu.be/-nrYB3PhK4M

johnoly99, 2016-04-11:
Also, we've tested it and it works 100% fine when we save the macro as a gcode text file and 'run' just the gcode.  The issue is in the macro and how macro's are sent.  We noticed that when running the gcode file, your gcode sender in the terminal's unfiltered output was putting a G4 P1000 in between each line.  Maybe there's something there to the reason the macros aren't working???

larsbrubaker, 2016-04-11:
Hey John, thanks for the video. We are going through this now.

larsbrubaker, 2016-04-11:
We have a fix for this and will be pushing it through test ASAP. 
This will go out as a maintenance release to fix this bug.
Thanks.

closed 2016-04-11 by larsbrubaker

johnoly99, 2016-04-11:
Awesome man!   

Thanks a ton!