MatterHackers/MatterControl #668: Bridge Fan Speed is being Ignored.

jjjohnson4112, 2016-04-12:
Hey guys, 

I have a customer that is having issues with the bridge fan speed in MatterControl. The setting is being completely ignored and the layer fan is never cued when there is a bridge. It also appears that the Bridge Speed settings are being ignored as well. I have viewed the customers GCODE and have sliced and verified for myself. 

Here is a quick summary video: https://youtu.be/QZEnQFb0b08

unlimitedbacon, 2016-05-18:
Confirmed this is happening on master. Here are [example slice settings and a test STL](https://github.com/MatterHackers/MatterControl/files/269433/BridgeTest.zip). It seems to be using the infill speed for bridging. Also, in my case it actually did turn the fans on, but immediately turned them off on the next line.
```
; LAYER:25
M106 S255
M106 S127
G0 X-34.589 Y4.5
```

unlimitedbacon, 2016-05-18:
This other [related issue](https://github.com/MatterHackers/MatterControl/issues/348) is for the wrong speeds.

closed 2016-05-24 by johnlewin