MatterHackers/MatterControl #675: Git commit d1b49492b35ea6d329ddab864b3c993a215735f2 breaks build

clytle374, 2016-04-24:
Not even sure if this is where I should post this.

I'm on Gentoo and getting this built the first time was a bit(or a lot) of trouble. I updated to try to fix a functional problem. Running the binary doesn't seem like an option of this machine either.

Anyway here is the output of git bisect

cory@cory-laptop ~/MatterDevelop/MatterControl $ git bisect good

d1b49492b35ea6d329ddab864b3c993a215735f2 is the first bad commit

commit d1b49492b35ea6d329ddab864b3c993a215735f2

Author: Lars Brubaker <larsbrubaker@gmail.com>

Date: Fri Apr 22 15:05:21 2016 -0700

Better Restore button

:040000 040000 eaf61302d3664f5479931cf27b1cf1b74016e11f 29986328f1f5714ee84e2190bcc27201e5658f96 M SlicerConfiguration

:040000 040000 32913be245096e4e2ba7869d69718aeb3b40d5d2 49a82c943f8f692e674c77eccee6d1a4ccf5d082 M StaticData

:040000 040000 17cc2149d941efd6da498997396a54a76044f259 feb2b9daedb6af1aa6dff49c27dfa3b45512b48a M Submodules

The build fails after that commit like this, I think this is the part you need.

ueue/QueueDataWidget.cs(522,17): warning CS0219: The variable `saveAsWindow' is assigned but its value is never used

Queue/QueueDataWidget.cs(574,26): warning CS0219: The variable `exportToSDProcess' is assigned but its value is never used

Queue/QueueDataWidget.cs(579,23): warning CS0219: The variable `project' is assigned but its value is never used

Queue/QueueDataWidget.cs(771,12): warning CS0219: The variable `enabledState' is assigned but its value is never used

SlicerConfiguration/SliceSettingsWidget.cs(380,7): warning CS0162: Unreachable code detected

SlicerConfiguration/SliceSettingsWidget.cs(1102,19): error CS1501: No overload for method `Line' takes `4' arguments

/home/cory/MatterDevelop/MatterControl/Submodules/agg-sharp/agg/bin/Release//AGG.dll (Location of the symbol related to previous error)

SlicerConfiguration/SliceSettingsWidget.cs(1103,19): error CS1501: No overload for method `Line' takes `4' arguments

/home/cory/MatterDevelop/MatterControl/Submodules/agg-sharp/agg/bin/Release//AGG.dll (Location of the symbol related to previous error)

SlicerConfiguration/SliceSettingsWidget.cs(1118,35): error CS1502: The best overloaded method match for `MatterHackers.Agg.UI.Button.Button(string)' has some invalid arguments

/home/cory/MatterDevelop/MatterControl/Submodules/agg-sharp/Gui/bin/Release//MatterHackers.Agg.UI.dll (Location of the symbol related to previous error)

SlicerConfiguration/SliceSettingsWidget.cs(1118,46): error CS1503: Argument `#1' cannot convert `MatterHackers.Agg.UI.ButtonViewStates' expression to type `string'

Task "Csc" execution -- FAILED

Done building target "CoreCompile" in project "/home/cory/MatterDevelop/MatterControl/MatterControl.csproj".-- FAILED

Done building project "/home/cory/MatterDevelop/MatterControl/MatterControl.csproj".-- FAILED

Task "MSBuild" execution -- FAILED

Done building target "Build" in project "/home/cory/MatterDevelop/MatterControl/MatterControl.sln".-- FAILED

Done building project "/home/cory/MatterDevelop/MatterControl/MatterControl.sln".-- FAILED

Build FAILED.

jlewin, 2016-04-25:
Hi clytle374, thanks for the report. On our end I'm not seeing any issues building off of the master branch for either our private build servers or the Travis CI servers. Is there any chance your submodules are out of sync and can you run ```git submodule``` from your MatterControl repository and report back?

Thanks,

clytle374, 2016-04-25:
Thanks for the reply, I did the following 

`cory@cory-laptop` ~/MatterDevelop/MatterControl $ git submodule
 c2856bbe822cf8abb380fdb5432f80b122e424e5 Submodules/MatterSlice (heads/master)
+9148b51ae473995e727e5d980c458e6da6bbfb96 Submodules/agg-sharp (b4tests-440-g9148b51)
cory@cory-laptop ~/MatterDevelop/MatterControl $ git checkout master
M	Submodules/agg-sharp
Previous HEAD position was 2c0f2c9... Making the new settings ui stuff work Stubbing out api to have user settings Removed unneeded class that held index
Switched to branch 'master'
Your branch is up-to-date with 'origin/master'.
cory@cory-laptop ~/MatterDevelop/MatterControl $ git submodule
 c2856bbe822cf8abb380fdb5432f80b122e424e5 Submodules/MatterSlice (heads/master)
+9148b51ae473995e727e5d980c458e6da6bbfb96 Submodules/agg-sharp (b4tests-440-g9148b51)
`

It still fails to build. I know gentoo is a bit ugly to everyone not running it. Here is the output, not sure if it's any different than before but I think it got truncated last time. thanks again

`SlicerConfiguration/SliceSettingsWidget.cs(380,7): warning CS0162: Unreachable code detected
SlicerConfiguration/SliceSettingsWidget.cs(1102,19): error CS1501: No overload for method `Line' takes `4' arguments
				/home/cory/MatterDevelop/MatterControl/Submodules/agg-sharp/agg/bin/Release//AGG.dll (Location of the symbol related to previous error)
SlicerConfiguration/SliceSettingsWidget.cs(1103,19): error CS1501: No overload for method `Line' takes `4' arguments
				/home/cory/MatterDevelop/MatterControl/Submodules/agg-sharp/agg/bin/Release//AGG.dll (Location of the symbol related to previous error)
SlicerConfiguration/SliceSettingsWidget.cs(1118,35): error CS1502: The best overloaded method match for `MatterHackers.Agg.UI.Button.Button(string)' has some invalid arguments
				/home/cory/MatterDevelop/MatterControl/Submodules/agg-sharp/Gui/bin/Release//MatterHackers.Agg.UI.dll (Location of the symbol related to previous error)
SlicerConfiguration/SliceSettingsWidget.cs(1118,46): error CS1503: Argument `#1' cannot convert `MatterHackers.Agg.UI.ButtonViewStates' expression to type `string'
			Task "Csc" execution -- FAILED
			Done building target "CoreCompile" in project "/home/cory/MatterDevelop/MatterControl/MatterControl.csproj".-- FAILED
		Done building project "/home/cory/MatterDevelop/MatterControl/MatterControl.csproj".-- FAILED
	Task "MSBuild" execution -- FAILED
	Done building target "Build" in project "/home/cory/MatterDevelop/MatterControl/MatterControl.sln".-- FAILED
Done building project "/home/cory/MatterDevelop/MatterControl/MatterControl.sln".-- FAILED

Build FAILED.`

jlewin, 2016-04-25:
Thanks for the quick response. From what I'm seeing it looks the problem stems from how we're using submodule branch tracking to minimize the noise and overhead submodule updates. Internally we sync everything up with ```"git submodule update --remote"``` from the MatterControl repository to push each submodule to their current master branch commit. Can you try running the submodule update command with the --remote param and see if it resolves the compile errors and if not, the ```"git submodule"``` output would again be helpful for continued troubleshooting.

clytle374, 2016-04-25:
Here is the output. It builds, and runs, fine now. Thanks a lot for the quick help. 

cory@cory-laptop ~/MatterDevelop/MatterControl $ git submodule update --remote
From https://github.com/MatterHackers/MatterSlice
 * [new branch]      1.5.2      -> origin/1.5.2
 * [new branch]      scene_bundle -> origin/scene_bundle
Submodule path 'Submodules/agg-sharp': checked out 'eb9701bc665499fb8631baa8a3a1853f79774ff8'

Should I add this to my notes for updating from git?  I'll probably clone this into a folder to run from and keep a current git just for fun and testing.  I'm new to this mono build, and 3d
printing for that matter. Thanks again for the help.

closed 2016-04-25 by clytle374

jlewin, 2016-04-25:
Great to hear that worked and yes, I believe any documentation that discusses building from sources should recommend the ```"git submodule update --remote"``` command  as it's the most convenient way to sync everything up. I'll make sure to update our build instructions to include this detail.

I looked for a link to your Gentoo notes but didn't see a url mentioned in either GitHub issue. If you want another set of eyes on it, drop a link and we'll take a look. Also, it wasn't clear if you previously were using our build instructions but it would be interesting to compare the process you went through on Gentoo to what we recommend in http://wiki.mattercontrol.com/Building_MatterControl.

clytle374, 2016-04-26:
Well, my notes wouldn't get someone far. More for the xbuild line. It's a long story, I'll try and keep it short. Firstly, I just tried to run the binary and found that it does work. Gentoo has some issues with the 32 bit libs and especially the 32 bit nvidia drivers. And some libs aren't where other distros put them, plus the 32bit libs are now built from the sources with a flag.

I now am pretty sure that most of my problems came from my laptop has(had) dual Nvidia cards, plus the integrated Intel video. Long story short I had to pull the 2nd nvidia card to access the intel card. Before that something in Mono ( I assUme)  would fail, and not understanding Mono, i was lost and thought I was missing files from the mono log files.  It was fixed by installing bumblebee to handle switching between Nvidia and Intel, and the 2nd Nvidia card never worked in Linux anyway. Bumblebee seems to add a layer to the graphics system that fixed it. So I am not sure what else along the way got Mattercontrol to work. 

I found several different things online about building it.  I started with your link.
http://wiki.mattercontrol.com/Building_MatterControl
It skips the ln -s to the static directory, the release build, and copying the files out of the binary. Are they all needed?

Then I found 
https://github.com/MatterHackers/MatterControl

I found a dependency in here somewhere. 
https://github.com/au-phiware/gentoo-overlay/blob/master/media-gfx/MatterControl/MatterControl-1.2.3.ebuild

I had trouble with several dependencies, I started building MC almost 6 weeks ago, then my printer arrived and I switched to windows temporally. So my memory is quite rusty. I have a spare HD and have been thinking about doing a fresh install on it to try and nail down a couple of things with MC and freecad build, maybe even try and write ebuilds for them. 

I do have one question tho, is MatterSlice being built and updated as well? Or is it rolled in?

Thanks again for the help

jlewin, 2016-04-28:
Thanks for the clarification and details. I've been a bit swamped with tasks but still have looking into this on my radar for some future weekend project because I'd like to experience the build process and roadblocks. When I get to that point, I'll certainly jump back here and compare. 

Good to know that the guide is missing instructions for the symbolic link step. That's a recent requirement after we dropped some questionable path manipulation which was causing errors in the wild for some users. I'll update the document as soon as I can create an account on the new MediaWiki server.

MatterSlice is definitely under active development with significant fixes and features in the works but we're a small team and have to juggle priorities across projects.

Thanks as well for trying out MatterControl on a new platform and for doing the hard work to sort through the process.

clytle374, 2016-04-28:
Feel free to ask questions when you try it.  I noticed that the Linux binary is a version behind windows and was having another problem I wasn't sure if it was the printer or the slicer. It's the reason I was updating and started this thread.  I have switched to stand alone slic3r since it is already in the gentoo system and built and installed easily.

If I decide to do a new install, I'll try and keep good notes as to what I had to do.  Also, release64 build always broke, I have thought in the back of my mind it would probably fixed several snags in the process, yet I didn't know if it was supposed to work.