MatterHackers/MatterControl #702: Add sequential printing option to slice settings for Slic3r and Cura

unlimitedbacon, 2016-05-02:
Cura and Slic3r both support sequential printing (printing one item at a time when you have a plate of many items) however, this option is not exposed in the slice settings UI.

It would be nice to have this feature implemented in MatterSlice as well.

johnlewin, 2016-05-27:
Need a description of best practices for how to achieve this. How are other open source solutions doing this?

unlimitedbacon, 2016-06-02:
The option to enable this is `complete_objects = 1`.

It doesn't work when given multiple objects in a single STL, though. This is why its been hidden in the UI until now. Slic3r needs to be given an AMF file. Unfortunately, [slic3r does not support compressed AMFs](https://github.com/alexrj/Slic3r/issues/3167), which is what MatterControl creates.