MatterHackers/MatterControl #712: Can't add printer

robcross, 2016-05-04:
I am running Debian Testing. I've compiled MatterControl from source.

After clicking "Save & Continue" in the 3D Printer Setup dialog, nothing happens, the dialog box stays open and the program doesn't crash... I don't really know what else to say :)

I have attached a log of me opening the program, clicking add printer, choosing a random make/model, clicking save and continue, then breaking.
[log.txt](https://github.com/MatterHackers/MatterControl/files/249368/log.txt)

johnlewin, 2016-05-05:
The master branch is currently unstable as we work through some issues. For at least the next week I'd recommend you build from the 1.5.3 branch

robcross, 2016-05-06:
Thanks, this works great :)

closed 2016-05-06 by robcross