MatterHackers/MatterControl #773: Slic3r homing problem on Delta machines.

Fishbowlkraken, 2016-05-14:
Reproducible bug on any 1.5 MatterControl installation, as far as I know. When using Slic3r through MatterControl on the Rostock Max, and possibly other deltas, the sliced model will always display that after centering itself, it will immediately retract, move diagonally back and to the right, then start the print one radius width of the bed from where it started. This means that under default settings, the head moves off of the bed in trying to start the print, and if trying to compensate for this error, the head moves off the bed trying to center itself. At first I'd thought this was an issue with a leftover configuration setting from before the update I ran last night. As such, I created a "dummy" Rostock, then proceeded to try and slice a model using Slic3r on it, as a test. Sure enough, I ran into the same issues. 
[image]
[image]
[image]

These are screenshots from the incidents I've described. I asked someone on the #reprap channel on Freenode about this, and they attempted a similar test using a "dummy" Rostock, to find the same results, though unfortunately I do not have a screenshot of their attempt. 

While I like MatterSlice, and it is certainly serviceable, I would like to be able to use Slic3r as well. This may be a problem exclusive to the Linux version of the application, as both of us were Ubuntu users, but either way, this is a fairly big problem.

unlimitedbacon, 2016-06-02:
So I did some more investigation into this. Right now MatterControl is using the following options to describe the bed geometry in the config file passed to Slic3r.
```
bed_shape = circular
bed_size = 250,250
print_center = 0,0
```
Slic3r changed how each of these options works in 1.2.9. `bed_shape` now takes a list of points describing a polygon. This allows them to have arbitrary bed shapes. For example...
```
bed_shape = 114.37x12.0208,112.487x23.9098,109.371x35.537,105.058x46.7747,99.5929x57.5,93.037x67.5953,85.4617x76.95,76.95x85.4617,67.5953x93.037,57.5x99.5929,46.7747x105.058,35.537x109.371,23.9098x112.487,12.0208x114.37,0x115,-12.0208x114.37,-23.9098x112.487,-35.537x109.371,-46.7747x105.058,-57.5x99.5929,-67.5953x93.037,-76.95x85.4617,-85.4617x76.95,-93.037x67.5953,-99.5929x57.5,-105.058x46.7747,-109.371x35.537,-112.487x23.9098,-114.37x12.0208,-115x0,-114.37x-12.0208,-112.487x-23.9098,-109.371x-35.537,-105.058x-46.7747,-99.5929x-57.5,-93.037x-67.5953,-85.4617x-76.95,-76.95x-85.4617,-67.5953x-93.037,-57.5x-99.5929,-46.7747x-105.058,-35.537x-109.371,-23.9098x-112.487,-12.0208x-114.37,0x-115,12.0208x-114.37,23.9098x-112.487,35.537x-109.371,46.7747x-105.058,57.5x-99.5929,67.5953x-93.037,76.95x-85.4617,85.4617x-76.95,93.037x-67.5953,99.5929x-57.5,105.058x-46.7747,109.371x-35.537,112.487x-23.9098,114.37x-12.0208,115x0
```
`bed_size` is a legacy option that is taken if `bed_shape` is not given. `print_center` is ignored completely. Since `bed_size` still works, there is no problem for cartesian printers. However, since `print_center` is now being ignored we have this problem on deltas.

unlimitedbacon, 2017-01-06:
Have another person on the forum running into this problem.
http://forums.matterhackers.com/topic/420/position-allways-off-after-generating-gcode

unlimitedbacon, 2017-01-25:
It appears that is issue is also preventing config files generated by MatterControl from being loaded into the Slic3r GUI. Ran into this problem while trying to investigate https://github.com/MatterHackers/MatterControl/issues/1821. Slic3r shows the following error.
```
I was unable to load the selected config file: illegal division by zero at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r/GUI/Plater/2D.pm line 256.
```
Looking at the Slic3r code briefly, [2D.pm line 256](https://github.com/alexrj/Slic3r/blob/1.2.9/lib/Slic3r/GUI/Plater/2D.pm#L256) seems to be part of a function related to setting the size of the bed. This makes me strongly suspect that the error is related to the fact that we are not telling Slic3r the bed size correctly.

closed 2017-01-30 by larsbrubaker

winter89, 2017-02-07:
I am having the same issue as described in this thread. Now that the issue has been marked as closed I would assume that there is a solution in place.

Is there information on how to go into MatterControl and modify the Slic3r code correctly?

unlimitedbacon, 2017-02-07:
The fix is in the latest source code but has not been released yet. You can wait for an alpha build to be made (probably in a few weeks) or download and compile the source if you want to try it out now.