MatterHackers/MatterControl #775: v 1.5.2 wont start up on Mac OS 10.11.4

jstefanop, 2016-05-16:
Matter Control never launches after it is installed on Mac OS 10.11.4 and becomes unresponsive...the process gets stuck at 100% CPU and need a force quit.

johnlewin, 2016-05-18:
Thanks for the feedback. We had an existing issue open in our old tracker that I've migrated to GitHub (#788) in order to better track the problem. 

It was believed that this Mono regression had been resolved in the 4.2.x releases that we ship in the app package. It's plausible that you might have an older Mono framework installed that's contributing to the issue and if so, I'd recommend removing any system installed Mono 4.x and trying with just the bundled app package version.

larsbrubaker, 2016-05-27:
dupe #788

closed 2016-05-27 by larsbrubaker