MatterHackers/MatterControl #780: GCode pipeline bypassed on back to back temperature statements

jlewin, 2016-05-17:
**Repro**
Print the following GCode file on an emulator board (a variant of this occurs in all dual extrusion prints)
```
M109 T0 S130
M109 T0 S130
```
**Expected**
All M109 statements should be replaced with M104s by the pipeline processor

**Actual**
The second M109 is not replaced with an M104 and the M109 passes through to  the printer

**Details**
The second M109 shortcuts the GCode pipeline when the WaitForTempStream reads and returns the next line without processing it, when it switches from `waitingForExtruderTemp` to `passthrough` mode. 

**Proposed Solution**
The stream should simply return null to allow the next call to naturally pass through the pipeline and avoid the short-circuiting

johnlewin, 2016-05-17:
Returning null is invalid as it indicates the stream has concluded and results in an immediate print finish. Returning an empty string fixes the problem

larsbrubaker, 2017-12-03:
This looks to be fixed with the referenced pull request. Closing.

closed 2017-12-03 by larsbrubaker