MatterHackers/MatterControl #787: Outer perimeters are sometimes printed first

unlimitedbacon, 2016-05-18:
When different sections of a print are linked only by a thin wall, the outer perimeters get combined. This causes the outer perimeter of some of the sections to get printed before the inner perimeters.

[image]

[Example settings and an STL](https://github.com/MatterHackers/MatterControl/files/271117/Perimeters-Issue.zip)

larsbrubaker, 2016-05-18:
Have a plan for doing this by checking a single vertex in each inner perimeter to see if it is inside, if so, computer inner before starting outer.

larsbrubaker, 2018-09-24:
Closing as fixed.
Please file a new bug if this is still a problem.

closed 2018-09-24 by larsbrubaker