MatterHackers/MatterControl #788: MatterControl hangs on Mac or Linux for minutes at a time

johnlewin, 2016-05-18:
This problem occurs somewhat randomly across similar machine and mono configurations but appears to be rooted in an underlying Mono regression related to System.Drawing and their FontConfig/font cache implementation.

This regression first appeared in Mono 4.0.4 and relates to accessing System.Drawing objects:

- https://bugzilla.xamarin.com/show_bug.cgi?id=17225
- https://forums.xamarin.com/discussion/9675/os-x-10-9-stuck-on-system-drawing-bitmap-ctor-calls

In the past we were forced to revert back to Mono 3.12.1 from the March 2015 timeframe but recent Mono build seem to have resolved the issue

johnlewin, 2016-05-18:
**Related Discussions**
https://forums.xamarin.com/discussion/8961/fontconfig-error-cannot-load-default-config-file
https://forums.xamarin.com/discussion/42593/how-to-work-around-absolute-fontconfig-path-in-mono-4-that-breaks-sandboxed-apps

jstefanop, 2016-06-03:
Does this have to do with horribly slow slicer performance on mac? For example this STL files takes about 15 minutes to slice...

http://www.thingiverse.com/thing:550604

rytz, 2016-09-27:
Support has received a recent influx of user emails since release of OS X Sierra. This is one of the top bugs for Mac, so several users will thank you for looking into it for the 1.6 build. :)

larsbrubaker, 2018-09-24:
Closing as fixed in 2.0.
Please file a new bug if this is still a problem.

closed 2018-09-24 by larsbrubaker