MatterHackers/MatterControl #813: Confusion over missing slice engine options with dual extrusion printers

unlimitedbacon, 2016-05-27:
Right now, MatterControl hides Cura and Slic3r from the list of slice engines when a dual extrusion machine is selected. This creates confusion because people don't know why those options are missing. Usually it leads people to believe that the other engines are not installed correctly.

### Suggested solution
Show Slic3r and Cura in the list, but have them grayed out so they cannot be selected. Also have a tooltip explaining why they cannot be used (MatterControl does not support dual extrusion using those engines).

unlimitedbacon, 2016-07-05:
[Help Scout ticket #996](https://secure.helpscout.net/conversation/222597823/0/?folderId=922212)

unlimitedbacon, 2016-07-07:
[Related forum thread](http://forums.matterhackers.com/topic/276/slicer-selection-issue)

unlimitedbacon, 2017-02-23:
[Another forum thread about this issue](http://forums.matterhackers.com/topic/473/slicer-engines-in-mattercontrol)

larsbrubaker, 2017-12-03:
2.0 only has MatterSlice. This is not an issue.

closed 2017-12-03 by larsbrubaker