MatterHackers/MatterControl #819: Cannot connect to 3D printer via USB

kcleung, 2016-05-29:
I am running on Ubuntu 16.04 and compiled the git master head ea5a7b9ee4045793dd531195f60778e006c12a37

However, the program cannot connect to the 3D printer (Formbot), despite the released version MatterControl 1.5 has no problems in doing so.

unlimitedbacon, 2016-05-31:
My guess would be that you didn't compile the serial helper library. This is needed for connecting to 250000 baud printers.
https://github.com/MatterHackers/MatterControl#serial-helper

Did you do this? You might also need to copy `libSetSerial` and `libSetSerial.so` to `bin/Release/`, although I believe that should be handled automatically if you build again.

kcleung, 2016-06-01:
I did all that, including copy libSerSerial to libSeralSerial.so

but still says "Unable to communicate with printer"

unlimitedbacon, 2016-06-02:
The only other thing I can think of would be a permissions issue, but since you can connect with the stable release that can't be the case. @jlewin, do you have any ideas?

kcleung, 2016-06-11:
I have just updated to de4fa24 and rebuilt everything, but same problem occurs.

By the way, do you have ppa packages for the newest release, as well as master snapshots?  The current ubuntu ppa package is quite outdated - only 1.5.0 - though 1.5.0 ppa ubuntu package can see my printer.

jlewin, 2016-06-13:
The serial helper is only required if you're using baud rates greater than 115200 and you shouldn't have to copy the libSetSerial components to your output directory as the required SerialHelper/build.sh script copies them for you. If you run the snippet below, it will enable the workarounds to support baud rates greater than 115200

```
#Change to the SerialHelper directory
cd Submodules/agg-sharp/SerialPortCommunication/SerialHelper

#Run the build script
./build.sh
```

This will compile and copy libSetSerial and libSetSerial.so (we need both) to the Release and Debug output directories, which get invoked when connecting to the serial port.

Have you tried creating a new profile since this issue began? The master branch is changing how printer profiles are stored to better support syncing across machines and at times the profile schema is in flux. Creating a new profile would ensure that nothing invalid exists in your current settings. Likewise, we've changed where you edit serial port assignment and baud rate and creating a new profile and resetting these fields is worth trying. 

If you're still unable to connect, feel free to attach a terminal log and I'll take a look

larsbrubaker, 2017-12-03:
Assume fixed. Please reopen with more details if not.

closed 2017-12-03 by larsbrubaker