MatterHackers/MatterControl #859: Migrate to new paid AppVeyor account

larsbrubaker, 2016-06-06:


johnlewin, 2016-06-27:
After working with AppVeyor it was determined that the paid accounts run on faster quad core virtual machines and use a completely different cloud architecture (GCE versus custom Hyper-V). 

To continue troubleshooting this issue we need to investigate command line nunit test failures that are easy to replicate on my machines and are likely the cause of the Hyper-V failures.

johnlewin, 2016-07-19:
Blocked by unstable automation tests that fail when run under the faster VM environment of the paid account

johnlewin, 2016-07-21:
Possibly fixed with Appveyor migration to Rackspace. I noticed the /larsbrubaker/ AppVeyor project is now building and passing tests. As such, I've deleted the /jlewin/ AppVeyor webhook from the MatterControl Github project and hope the behavior that is currently being exhibited is long lasting

closed 2016-07-22 by larsbrubaker