MatterHackers/MatterControl #936: No Z lift on tool change retractions

unlimitedbacon, 2016-06-13:
Z lift is not being applied to tool change retractions. Here is an example. 

```
G1 F3600 X135.76 Y144.345 E50.44532
G1 F1800 E40.4453 ; retract
G92 E0 ; reset extrusion
T1 ; switch extruder
G0 F9000 X149.218 Y122.6
; TYPE:SUPPORT
G1 Z5.3
G1 F1800 E10
G1 F2400 X149.6 Y120.941 E10.04247
```

It looks like it has the drop down command (`G1 Z5.3`), but not the lift up command. Z was already at 5.3 at this point.

larsbrubaker, 2016-06-13:
What is the expected behavior?

unlimitedbacon, 2016-06-13:
This is a regular (non-toolchange) retraction from the same file
```
G1 F3600 X86.192 Y145.497 E48.9846
G1 F1800 E45.9846
G1 Z5.8
G0 F9000 X88.689 Y145.406
...
G0 X129.371 Y143.945
G1 Z5.3
G1 F1800 E48.9846
G1 F3600 X143.184 Y130.133 E49.47187
```
So I believe what it should be doing is this:
```
G1 F3600 X135.76 Y144.345 E50.44532
G1 F1800 E40.4453 ; retract
G1 Z5.8
G92 E0 ; reset extrusion
T1 ; switch extruder
G0 F9000 X149.218 Y122.6
; TYPE:SUPPORT
G1 Z5.3
G1 F1800 E10
G1 F2400 X149.6 Y120.941 E10.04247
```