MatterHackers/MatterControl #947: Offset fields of the menu item Settings -> Printer -> Extruder do not specify X and Y offset

kcleung, 2016-06-15:
I am using MatterControl 1.5.3, and when I go to:
Settings -> Printer -> Extruder 
For each extruder, the offset has two fields, but do not specify which one is X and which one is Y.
Also suppose if I have the distance between the first and second extruder, then I enter this distance as the X offset of the second extruder, right?  If the second extruder is at the left of the first extruder. then do I put the offset as negative X?  These details are not specified in either wikipedia or the tooltip.

larsbrubaker, 2016-06-15:
The x y labels will be in the next version.
Yes you would put the second extruder offset in, and it can be negative.
We will update the wiki.

Thanks for the feedback.

kcleung, 2016-06-18:
When you say "next version" do you mean 1.5.4?
Also, is this fix already in the git master?
By the mean time, could you please tell me which one is the X and which one is the Y?

unlimitedbacon, 2016-06-20:
The next version will be 1.6. Yes, this is already fixed in the master branch. First box is X, second box is Y.
http://wiki.mattercontrol.com/Settings/Printer/Extruder/Extruder_X/Extruder_Offset